MCUdude/MajorCore #1: Seriously, I need a better name for this core!

MCUdude, 2016-10-24:
I chose the title 8051core as a working title, because the ATmega8515 and ATmega162 is pin compatible with the old 8051's. I don't feel like merging this into MightyCore, since these chips don't got the same pinout and peripherals. No ADC and i2c for instance.

If it's possible, I want to make the core start with the letter 'M', like all the other cores, but a suited name is more important than the first letter.

<b>A little background info:</b>
- The AT90S8515/ATmega8515 was one of the very first AVR microcontrollers released, and where (almost, except the reset pin) pin compatible with the old 8051.

- They have been around for a long time, and are considered "mature"

- The ATmega8515 can still be bought for less than 1.50$ at Ebay

- The ATmega8515 was shipped with the STK500 starter kit, so a lot of people are unaware that they (now) own a capable Arduino compatible microcontroller.

Does "core master" @drazzy or "boards manager master" @per1234 have any great names they want to share? 😉

SpenceKonde, 2016-10-25:
AntiqCore? 

If one of us is the core master here, it is probably yourself - I'll note that I maintain two cores counting generously, one from the users' perspective. You've got mighty, mini, micro and this one, for four of em. Though I reckon the mini and mighty have way less weirdness than the 'tinys.

per1234, 2016-10-25:
Well I did some brainstorming on some possible names that start with "M". I'm not sure if any of these are especially fitting but if nothing else maybe will inspire other ideas:
- MantiCore - manticore is a mythological creature
- MesoCore - meso- prefix means middle or intermediate
- MacroCore
- MagniCore - magni- prefix means great
- MuCore - Mu is a greek letter μ used to indicate the prefix micro-

As far as a non-M name referring to the old school nature of the 8515, RetroCore or if you consider it really ancient then PaleoCore.

How about something with a Norwegian flavor to give tribute to the roots of AVR?

MCUdude, 2016-10-25:
Hmm.. Norwegian flavour? That's a hard one. The AVR was developed in the norwegian city Trondheim, which has a really old catherdal called Nidarosdomen. Still, I don't know if names like NidarosCore will fit, even though it's a great tribute to the AVR.

How about MajorCore? When the AVRs hit the marked in the late 90s, is was a major game changer bacause of its superb arcitecture and internal flash strorage (while others where using OTP ROMS and EPROMS)

per1234, 2016-10-25:
FjordCore? :smile:

MCUdude, 2016-10-26:
Thats the most touristy name ever 😆I don't know why, but a lot

I think I'll go for MajorCore. It starts with an m, and IMO the word "major" makes sense 👍 
But thanks for the brainstorming!

closed 2016-10-26 by MCUdude