MCUdude/MajorCore #2: Add boards manager support

MCUdude, 2016-11-08:
The core is almost finished, and I think it's ready for its first release. I'm not sure if I'm going to to create an Arduino port about this core, but I'm adding it to the "Arduino on other Atmel chips" page. Would you like to help @per1234 ? 👍

per1234, 2016-11-08:
Sure, will do it later tonight when I get home. Good work!

MCUdude, 2016-11-08:
Great, Thanks!  Good luck with the election today. Every Norwegian newspaper have been covered with US politics the last few weeks. We have an ongoing bet at work on who's getting elected. I'll become 50 NOK / 6$ richer if Hillary wins 😛

per1234, 2016-11-08:
Well I did my part to send that money your way. I sure do hope you'll win the bet!

closed 2016-11-09 by MCUdude