MCUdude/MajorCore #6: atmega162 problem

javest, 2017-05-29:
I tried to burn bootlader of atmega162 using your library and arduino as ISP with Arduino IDE 1.6.5.
I recevive the following error.
avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

I tried with internal or external oscillator.

I programmed successfully with the same method and wiring an atmega8515.
I managed to load bootlader and sketch. 

With the atmega162, neither of the two programming worked.

I tried with different chips of the same type, baut the beahavior is the same.

Could you help me, please?

MCUdude, 2017-05-29:
There's clearly a hardware issue here. you can see the device signature is 0x000000. This means that the MISO line is pulled to ground somehow. Is the ISP interface disabled on the ATmega162?

angel77777, 2017-05-29:
Hi,
i'm working with javest.
We've replaced the 8515 with a 162 in the same position on the breadboard and we've tried 2 different 162.
The 8515 works, the 162 not.
Miso,mosi,sck,reset,vcc,gnd from arduino uno to 162.
10k resistor on the 162 from reset and vcc
10uF capacitor on the Uno between reset and gnd.

We've tried with 16mhz cristal +22pF on the 162 anche without it.
I'll test the miso (it is not connected to gnd and anyway the 8515 works).
Thanks..

MCUdude, 2017-05-29:
There might be a bug in the avrdude.conf file. To be honest, I haven't really tested it on the ATmega162 because I don't have one. Since it works great with ATmega8515, and builds just fine on ATmega162, I thought it was ok.

Please enable verbose output in the IDE. when you try to burn the bootloader, a command will show up in white text. Use that avrdude path, and run this command from your command line interface (terminal, cmd, PS) `/path/to/avrdude -v -p atmega162 -c stk500v1 -P COMx -b19200`

Obviously you need to replace COMx with your com port (assuming you're using windows). If you report back soon I might be able to fix this tonight!

angel77777, 2017-05-29:
this is what the verbose output writes:

avrdude: Version 6.3, compiled on Dec 16 2016 at 13:32:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/samuelelini/Documents/Arduino/hardware/MajorCore-master/avr/avrdude.conf"
         User configuration file is "/Users/samuelelini/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem1411
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega162
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash         65    10   128    0 yes     16384  128    128  4500  4500 0xff 0xff
           eeprom        65    20     4    0 no        512    4      0  9000  9000 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0 16000 16000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0 16000 16000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0 16000 16000 0x00 0x00
           lock           0     0     0    0 no          1    0      0 16000 16000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x000000 (retrying)

Errore durante la scrittura del bootloader
Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

MCUdude, 2017-05-29:
Thanks. I need a little more information though. I need the actual command issued. at my computer it's:
`/Users/username/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/Users/username/Documents/Arduino/hardware/MajorCore/avr/avrdude.conf -v -patmega162 -cstk500v1 -P/dev/cu.wchusbserial1410 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xfb:m -Uhfuse:w:0xd4:m -Ulfuse:w:0xf7:m `

angel77777, 2017-05-29:
ok, this is the "white" line:

private/var/folders/p0/6jt_7ns55hb2y2w5vm5kyct80000gn/T/AppTranslocation/5F671BA8-E2DF-44D8-9344-9ECE5F9F2EFA/d/Arduino 1_8_1.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/samuelelini/Documents/Arduino/hardware/MajorCore-master/avr/avrdude.conf -v -patmega162 -cstk500v1 -P/dev/cu.usbmodem1411 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xd4:m -Ulfuse:w:0x62:m

MCUdude, 2017-05-29:
Great, you're using a mac; me too 🙂 

Now open Terminal and paste this exact command:
`/private/var/folders/p0/6jt_7ns55hb2y2w5vm5kyct80000gn/T/AppTranslocation/5F671BA8-E2DF-44D8-9344-9ECE5F9F2EFA/d/Arduino 1_8_1.app/Contents/Java/hardware/tools/avr/bin/avrdude  -v -patmega162 -cstk500v1 -P/dev/cu.usbmodem1411 -b19200`

and post the output. What I've done is removed the avrdude.conf configuration from the command. I suspect there might be something wrong in there.

angel77777, 2017-05-29:
-bash: /private/var/folders/p0/6jt_7ns55hb2y2w5vm5kyct80000gn/T/AppTranslocation/5F671BA8-E2DF-44D8-9344-9ECE5F9F2EFA/d/Arduino: No such file or directory

MCUdude, 2017-05-29:
Same result if you remove the slash in front of "private"? 

Have you installed the IDE property by dragging it into the Applications folder?

per1234, 2017-05-29:
In the Windows world we need to put quotes around:
```
private/var/folders/p0/6jt_7ns55hb2y2w5vm5kyct80000gn/T/AppTranslocation/5F671BA8-E2DF-44D8-9344-9ECE5F9F2EFA/d/Arduino 1_8_1.app/Contents/Java/hardware/tools/avr/bin/avrdude
```
The space in `Arduino 1_8_1.app` causes just the sort of error that @angel77777 reported.

MCUdude, 2017-05-29:
That's correct. Put a backslash right in front of the space there.

/private/var/folders/p0/6jt_7ns55hb2y2w5vm5kyct80000gn/T/AppTranslocation/5F671BA8-E2DF-44D8-9344-9ECE5F9F2EFA/d/Arduino\ 1_8_1.app/Contents/Java/hardware/tools/avr/bin/avrdude -v -patmega162 -cstk500v1 -P/dev/cu.usbmodem1411 -b19200

angel77777, 2017-05-29:
same error:

-bash: /private/var/folders/p0/6jt_7ns55hb2y2w5vm5kyct80000gn/T/AppTranslocation/5F671BA8-E2DF-44D8-9344-9ECE5F9F2EFA/d/Arduino 1_8_1.app/Contents/Java/hardware/tools/avr/bin/avrdude: No such file or directory

MCUdude, 2017-05-29:
It seems like your Arduino install is stored in a temporary folder. By moving Arduino.app to applications, you'll get a known path to avrdude, which we're struggling with now.

angel77777, 2017-05-29:
how can i install DE property by dragging it into the Applications folder?

I've always been using arduino ide without problem and the 8515 works well

angel77777, 2017-05-29:
ok, i try

angel77777, 2017-05-29:
this is the "white" text after i've dragged arduino 1_8_1.app to application folder:

Applications/Arduino 1_8_1.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/samuelelini/Documents/Arduino/hardware/MajorCore-master/avr/avrdude.conf -v -patmega162 -cstk500v1 -P/dev/cu.usbmodem1411 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xd4:m -Ulfuse:w:0x62:m

MCUdude, 2017-05-29:
Much better!

Now paste this command in your Terminal:

`Applications/Arduino\ 1_8_1.app/Contents/Java/hardware/tools/avr/bin/avrdude  -v -patmega162 -cstk500v1 -P/dev/cu.usbmodem1411 -b19200`

and post the output

angel77777, 2017-05-29:
-bash: Applications/Arduino 1_8_1.app/Contents/Java/hardware/tools/avr/bin/avrdude: No such file or directory

MCUdude, 2017-05-29:
Sorry, forgot a slash there..

`/Applications/Arduino\ 1_8_1.app/Contents/Java/hardware/tools/avr/bin/avrdude -v -patmega162 -cstk500v1 -P/dev/cu.usbmodem1411 -b19200`

This should do it!

angel77777, 2017-05-29:
avrdude: Version 6.3, compiled on Dec 16 2016 at 13:32:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/jenkins/jenkins/workspace/avrdude/label/mac-mini/objdir/etc/avrdude.conf"
avrdude: can't open config file "/Users/jenkins/jenkins/workspace/avrdude/label/mac-mini/objdir/etc/avrdude.conf": No such file or directory
avrdude: error reading system wide configuration file "/Users/jenkins/jenkins/workspace/avrdude/label/mac-mini/objdir/etc/avrdude.conf"
Mac-mini-di-Samuele:~ samuelelini$

MCUdude, 2017-05-29:
OK, so it needs a config file.. Try this

`/Applications/Arduino\ 1_8_1.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Applications/Arduino\ 1_8_1.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -v -patmega162 -cstk500v1 -P/dev/cu.usbmodem1411 -b19200`

angel77777, 2017-05-29:
avrdude: Version 6.3, compiled on Dec 16 2016 at 13:32:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino 1_8_1.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/samuelelini/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem1411
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega162
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash         65    10   128    0 yes     16384  128    128  4500  4500 0xff 0xff
           eeprom        65    20     4    0 no        512    4      0  9000  9000 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0 16000 16000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0 16000 16000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0 16000 16000 0x00 0x00
           lock           0     0     0    0 no          1    0      0 16000 16000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

angel77777, 2017-05-29:
Problem is here right?

User configuration file is "/Users/samuelelini/.avrduderc"
 User configuration file does not exist or is not a regular file, skipping

Maybe wrong installation?

MCUdude, 2017-05-29:
Ok, it's what I feared. There isn't anything wrong with MajorCore at all.

What we're done now is that we're ran avrdude (used for setting fuses + uploading) directly from the command line, and excluded every reference to MajorCore. It still didn't work.

It may seem weird, but it looks like to be hardware related. Are those ATmega162's brand new, or pulled from working equipment? If they aren't brand new, they a) have their ISP interface disabled and/or b) programmed to use an external oscillator. Did you add two 22pF from the external crystal to ground? I've been using my trusty old signal generator (5v square wave, 1 MHz or more) when I've messed up the oscillator settings. Apply a square wave signal to pin XTAL1, and you're manually driving its clock. It _may_ be the problem after all..

> Problem is here right?
User configuration file is "/Users/samuelelini/.avrduderc"
User configuration file does not exist or is not a regular file, skipping
Maybe wrong installation?

No, it's a regular warning and does not affect the result

I'm afraid I can't do any testing myself because I don't have an ATmega162 at hand..
I need to go to bed now. I'll be back tomorrow 👍

angel77777, 2017-05-29:
The 162 are new and i'm trying to use external oscillator with the 22pF too to gnd..
I don't know if the productor can disable the isp (8515 came frome the same "house" and have not this problem).

Ok, thanks a lot,
see you tomorrow! :)

angel77777, 2017-05-29:
ps: here you can find some pictures of my wiring:

https://www.dropbox.com/sh/vmlhhnw733wus9k/AAATRj65O7kGkzySeWBZOfeDa?dl=0

angel77777, 2017-05-30:
hi,
i've put 3v square wave, 1Mhz and 8Mhz(my generator can reach only 3V) on XTAL1 with no success (i've tried to upload bootloader and sketch with external clock). Anyway i have a 16mhz crystal with the 22pF and it doesn't work..
Maybe the problem is the arduino UNO? (quite weird as the 8515 works well).

There something you can get from this doc?
http://www.atmel.com/Images/doc2536.pdf
thanks..

MCUdude, 2017-05-30:
There's nothing in the migration document you need to take into account. What you can try to do is to add a 100nF capacitor as close to the Vcc and GND pin on the chip as possible. It's highly recommended, and some have reported that their chip doesn't work propery withou this cap.

Even though the ATmega8515 and ATmega162 is pin compatible, they may have internal differences that makes the 8515 work without a capacitor, and the 162 not.

Do you have anything else that a breadboard to test on? You can use an old STK500 board if you can find one.

angel77777, 2017-05-30:
I've already added the 100uF capacitor near Vcc, i'll try it again.
Very soon i'll can try a STK500 to do the work.
Unlukely at the moment i can try it just on a breadboard..
thanks again!

angel77777, 2017-06-01:
hi,
we can be more precise: if i upload the sketch at 1mhz or 8mhz(internal) when the chip is new form factory it works.
It doesn't work no more after i've tried to upload the bootloader at 16mhz with external (16mhz)crystal and 22pF capacitor.
After that is not possible anymore to restore, from arduino IDE,  factory settings or upload anything.

MCUdude, 2017-06-01:
.. So if you're only using the internal oscillator, it works just fine. After you're selected 16 MHz, you cannot communicate with the cip again, right?

MCUdude, 2017-06-01:
I think I found the problem. It's the low fuse.. Try changing the low fuse in the boards.txt file to 0xFF instead. This will activate the oscillator driving circuit inside the ATmega162, which will drive the external crystal

angel77777, 2017-06-01:
"So if you're only using the internal oscillator, it works just fine"
Yes, right!
We changed the board.txt file but because the error doesn't change we suppose that the fuse were already burned wrong on the 162.
So how can we reset it to factory settings?

MCUdude, 2017-06-01:
If you have access to an STK500 board, you can use its high voltage parallel programming features to bypass everything. If not, you'll have to apply a clock signal into XTAL1.

angel77777, 2017-06-01:
so you mean an external 16mhz clock generated with an oscilloscope?
At the moment we only have a 16mhz xtal and an oscilloscope that reach 8mhz maximum..

angel77777, 2017-06-01:
I think with our equipment we cannot restore it so:
are you sure now that modifying the fuse settings it will work on a new 162?
You plan to try yourself on a 162?

we want to know if we have got to buy a new 162 or a HVPP to restore the ones with wrong fuse..

MCUdude, 2017-06-01:
> so you mean an external 16mhz clock generated with an oscilloscope?
At the moment we only have a 16mhz xtal and an oscilloscope that reach 8mhz maximum..

An oscilloscope itself cannot generate any signal. You need a function generator for that. When trying to restore an AVR microcontroller, the clock frequency you apply to it doesn't really matter. It just have to be at least 4 times faster than the programmer SCK signal. This is usually 250kHz or lower, so a 1 MHz clock signal or greater works just fine. 

Try uploading this code to another Arduino. This will toggle PD0 (digital pin 0 on Arduino UNO) with a speed of a couple of megahertz. Connect PD0 to XTAL1 on the ATmega162. This way you should be able to communicate with it again.

``` c
void setup() 
{
  DDRD |= _BV(PD0); // Set PD0 as output
}

void loop() 
{
  PORTD |= _BV(PD0); // Set PD0 high
  PORTD &= ~_BV(PD0); // Set PD0 low
}
```

The fuse settings have disabled the crystal driver, so in order to communicate with it, you need to apply a signal from a clock source. This can actually be another Arduino running a loop while toggling a pin, or it can be a dedicated function generator. This is by far the simplest solution to fix this. 

The next option is to build a HVPP. I did this once when I screwed up the fuse settings on my UNO. I used [this guide](http://mightyohm.com/blog/products/hv-rescue-shield-2-x/) for building one. You need a 12v source though.

angel77777, 2017-06-01:
thanks a lot, you've been very clear, i was on the same way :)
My oscilloscope can generate square function too at 1mhz ;) ..but it seems it was not working as the output of the square wave was less than 3V.
If the first method will not work i'll try to build the HVPP.
I'l try to connect the pins on the 162 to the pins with the same name as the ones on the 168/328 series(your guide do not talks about 40pin mcus), i hope they are the same :)

MCUdude, 2017-06-01:
I've looked at the HVPP sketch I linked to, and it not compatible any 40-pin devices. I'm putting my bet on the Arduino signal generator :)

angel77777, 2017-06-01:
unluckely there is no way:
-arduino UNO as isp
-arduino MEGA(with your sketch) as clock source wired from PD0 (pin 21 on MEGA) and gnd to  xtal1 and gnd on 162.
Are you planning to purchase an atmega162 or can you confirm the new fuse settings in board.txt are right?
thanks!

MCUdude, 2017-06-01:
> -arduino MEGA(with your sketch) as clock source wired from PD0 (pin 21 on MEGA) and gnd to xtal1 and gnd on 162.

If you're looking at an Arduino MEGA pinout diagrams, you can see that PD0 is actually assigned to digital pin 21. Try connecting digital pin 21 to XTAL1 🙂  

I've updated the boards.txt file on github with new values. I don't have any hardware to test on, but I'm 100% sure the fuse values are correct now. I'm using the same setup for MightyCore. The ATmega162 is very expensive here in Norway, and you can't get a cheap DIP package from Ebay either.

angel77777, 2017-06-03:
The square wave was not a square so i've modified the code with some "no operation instructions" to be more like a wave but it doesn't work.
At the moment i'm trying HVPP at the link you've sent to me as he says " This program should work for other members of the AVR family, but has only been verified to work with the ATmega168".
I'll let you know..

angel77777, 2017-06-11:
I confirm that now the 162 works with the updated fuse at 16mhz.
The hvpp didn't work for me so i'll fix the "broken" 162 with an oscilloscope on xtal1 as i'll can.
thanks for support!

MCUdude, 2017-06-11:
Great to hear. I'm sorry about the wrong fuses, but on the bright side; it's fixed now. Thank you for reporting the issue and for using MajorCore 😃

closed 2019-03-18 by MCUdude