MCUdude/MajorCore #10: Optiboot  ..undefined reference to 'eeprom_write_byte'

zvipesh, 2017-07-30:
I need to create a custom bootloader for mega1284P with 14.746MHz , baudrate=57600
and LED on port C7.

I am using the C:\arduino183\hardware\MightyCore\avr\bootloaders\optiboot.old database.
changed the OMAKE.bat path to C:\WinAVR-20100110\utils\bin\make OS=windows ENV=arduino %*
and compiling the new entry.

I get error messages : 
Optiboot  ..undefined reference to 'eeprom_write_byte'
Optiboot  ..undefined reference to 'eeprom_read_byte'
see attached :

in optiboot.c  the code shows 
/*
 * void writebuffer(memtype, buffer, address, length)
 */
static inline void writebuffer(int8_t memtype, uint8_t *mybuff,
			       uint16_t address, pagelen_t len)
{
    switch (memtype) {
    case 'E': // EEPROM
#if defined(SUPPORT_EEPROM) || defined _(BIGBOOT)_ 
        while(len--) {
	    eeprom_write_byte((uint8_t *)(address++), *mybuff++);
        }
#else
	/*
	 * On systems where EEPROM write is not supported, just busy-loop
	 * until the WDT expires, which will eventually cause an error on
	 * host system (which is what it should do.)
	 */
same for the READ_BYTE section.

it looks like the BIGBOOT parameter is used only in this eeprom section.
I am NOt using the EEPROM ... so what is the solution ?
I was thinking of just changing the code to :
  
  switch (memtype) {
    case 'E': // EEPROM
#if defined(SUPPORT_EEPROM) 
        while(len--) {
	    eeprom_write_byte((uint8_t *)(address++), *mybuff++);
        }
#else

basically removing the BIGBOOT  parameter.

please advise if this is OK  ... and also what is the syntax for NOT using EEPROM 
if using the optiboot_flash code ?

Link to screen shots and source code  is located at : 
 https://drive.google.com/open?id=0B8APtZDn4cUtS3RXYm1rUUtnc1U

thanks 
zvipesh@gmail.com

MCUdude, 2017-07-30:
This repo does not support Atmega1284, only Atmega8515 and 162. 

Seems like the linker cannot find eeprom.h.
Why are you using the optiboot.old version? Optiboot_flash is better, under active development and compiles without any issues. I do not provide any support for the legacy bootloader anymore, so try that one if you want something that works.

zvipesh, 2017-07-30:
Thanks Hans,

I tried first the Optiboot_Flash .. got the same errors ...so reverted back
to optiboot_old ...

went back to optiboot_flash
deleted  || ( BIGBOOT ) from EEPROM statement   and it compiles OK with no
errors .

question:  where is the documentation as to the command structure when
wanting to write to EEPROM ?

thx

MCUdude, 2017-07-31:
I see you've compiled used WinAVR from 2010, which is quite old. The default location for the eeprom.h file is avr/eeprom.h. It may be different in older versions of avr-libc. Anyways I recommend compiling with avr-gcc 4.9.x and the newest avr-libc. A lot have happened since 2010.

If you're not using the BIGBOOT option, you're safe to remove the BOGBOOT statement to get rid of the EEPROM errors

zvipesh, 2017-08-01:
Hi Hans:

thanks for your response.

I did removed the BIGBOOT and it compiled OK.

I modified the package installed to include my custom board definitions.
and added a bootloader variant.

I just modified the boards.txt to include my 14.746MHz and 12MHZ with
Resonators.
that does not work properly .. for what ever reason I am getting errors.

I selected the  "standard"  1284P ( original code )  and I do not get the
errors ...

loaded Blink and it worked.

However ...

I can only load a new sketch  after I load the bootloader !
once I loaded a sketch  the 1284P does not recognize the serial port
attempts to load new sketch.

I looked at the code and it looks like the bootloader is sitting in a for
(;;;) loop and looking for character from the serial port ... but in my
case it does not get it for the second
sketch.

I had a similar problem with the AT2560 using the STK500V2 bootloader.
I had to play with the delay to make almost 8 sec where the bootloader is
looking for serial input.

It may be the same problem  but I do not see anywhere a timer / delay
function :

  /* Forever loop: exits by causing WDT reset */
  for (;;) {
    /* get character from UART */
    ch = getch();

    if(ch == STK_GET_PARAMETER) {
      unsigned char which = getch();
      verifySpace();
      /*
       * Send optiboot version as "SW version"
       * Note that the references to memory are optimized away.
       */
      if (which == 0x82) {
 putch(optiboot_version & 0xFF);
      } else if (which == 0x81) {
 putch(optiboot_version >> 8);
      } else {
/*
* GET PARAMETER returns a generic 0x03 reply for
         * other parameters - enough to keep Avrdude happy
*/
putch(0x03);
      }
    }
    else if(ch == STK_SET_DEVICE) {
      // SET DEVICE is ignored
      getNch(20);
    }
    else if(ch == STK_SET_DEVICE_EXT) {
      // SET DEVICE EXT is ignored
      getNch(5);
    }
    else if(ch == STK_LOAD_ADDRESS) {
      // LOAD ADDRESS
      uint16_t newAddress;
      newAddress = getch();
      newAddress = (newAddress & 0xff) | (getch() << 8);
#ifdef RAMPZ
      // Transfer top bit to RAMPZ
      RAMPZ = (newAddress & 0x8000) ? 1 : 0;
#endif
      newAddress += newAddress; // Convert from word address to byte address
      address = newAddress;
      verifySpace();
    }
    else if(ch == STK_UNIVERSAL) {
      // UNIVERSAL command is ignored
      getNch(4);
      putch(0x00);
    }
    /* Write memory, length is big endian and is in bytes */
    else if(ch == STK_PROG_PAGE) {
      // PROGRAM PAGE - we support flash programming only, not EEPROM
      uint8_t desttype;
      uint8_t *bufPtr;
      pagelen_t savelength;

      GETLENGTH(length);
      savelength = length;
      desttype = getch();

      // read a page worth of contents
      bufPtr = buff;
      do *bufPtr++ = getch();
      while (--length);

      // Read command terminator, start reply
      verifySpace();

#ifdef VIRTUAL_BOOT_PARTITION
#if FLASHEND > 8192
/*
 * AVR with 4-byte ISR Vectors and "jmp"
 */
      if (address == 0) {
// This is the reset vector page. We need to live-patch the
// code so the bootloader runs first.
//
// Save jmp targets (for "Verify")
rstVect0_sav = buff[rstVect0];
rstVect1_sav = buff[rstVect1];
wdtVect0_sav = buff[wdtVect0];
wdtVect1_sav = buff[wdtVect1];

        // Move RESET jmp target to WDT vector
        buff[wdtVect0] = rstVect0_sav;
        buff[wdtVect1] = rstVect1_sav;

        // Add jump to bootloader at RESET vector
        buff[rstVect0] = ((uint16_t)main) & 0xFF;
        buff[rstVect1] = ((uint16_t)main) >> 8;
      }

#else
/*
 * AVR with 2-byte ISR Vectors and rjmp
 */
      if ((uint16_t)(void*)address == rstVect0) {

It looks like the bootloader is "waiting " for about 4 seconds before
jumping to the Sketch code
( from reset to  start of  LED flashes from the sketch code )

can you point out how to change the Boot timeout ??

the notes at the begining of the optiboot_flash include a comment:

/* TIMEOUT_MS:                                            */
/* Bootloader timeout period, in milliseconds.            */
/* 500,1000,2000,4000,8000 supported.                     */
/*                                                        */

but there is no where a call to 'TIMEOUT_MS' in the code .

thank you.

I would like to know if there is a version of the bootloader similar to the
ATmega2560 bootloader in the Arduino IDE

-->  C:\arduino183\hardware\arduino\avr\bootloaders\stk500v2
that is what I used for the mega2560 and it works ...

tried to use it for 1284P  and was getting a lot of errors related to
missing files etc.

thanks for your help

MCUdude, 2017-08-01:
Are you sure your audio reset circuity is OK? Have you added a 100nF capacitor between DTR on your serial adapter and RST on your microcontroller? If not, uploading will only work once. I'm no bootloader expert, so I'm afraid I want answer all your questions. The [Official Optiboot repo](https://github.com/Optiboot/optiboot) is the correct place 👍

zvipesh, 2017-08-01:
yes, added the 0.1uF capacitor between the CP2104 DTR pin and the Reset
line of the uP.

I saw several posts to the fact the 1284P is very "finicky"  so I also
added an external oscillator instead of the Resonator.
no change.

my biggest battle is that there is a different source for Bootloaders for
2560 and 1284 ...
most of the 1284 is related to the IDE 1.3 or 1.5 ... I am using the IDE
1.8.3
and you would think that the bootloaders will include the 1284P
under the bootloaders\STK500 folder.

as I mentioned before .. the bootloader for the 2560 shows the 1284 in
several places in the bootloader
but when compiling I get errors related to different files missing and
different paths and in short   it is a
"pain"

I need to get the bootloaders working and it takes forever to get an
straight answers from the people
involved in writing the bootloaders. ...

any help will be greatly appreciated.

thank you

MCUdude, 2018-06-05:
I'm closing this as this has mothing to do with ATmega8515 or ATmega162

closed 2018-06-05 by MCUdude