MCUdude/MajorCore #11: Bootloading ATMEGA8515L

RayyanTahir, 2018-05-26:
Hi, I'm new to bootloading. I Followed the instructions with the circuit provided in the details. After running ArduinoISP on ArduinoUno and bootloading to ATMEGA8515L at first I got a message like `Device Signature: 0x1e9301` whereas in the config the signature for the microcontroller was set as `0x1e9306` which led the bootloader to stop. I ran the bootloader a couple of times and after a few restarts of Arduino IDE and my PC, the device signature always shows as `0x00000000`. 

Your help will be appreciated. Thanks

RayyanTahir, 2018-05-27:
I tried adding another `ATMEGA8515L` but when I try to bootload it I get `Device Signature: 0x00ff00`. The second time I try to bootload it, it gives `Device Signature: 0xff0000` and then doing this repeatedly the `ff` part of the address just changes between like `0xff0000`, `0x00ff00` and `0x0000ff`.

MCUdude, 2018-05-27:
Hi!
This is most likely caused by an improper connection. Can you post a picture of your physical hardware? You should also check all connections, including the reset line

Eemilp, 2019-04-13:
Hi @RayyanTahir 

I'm far from an expert but had very similar issue and got it figured out. 

Assuming your wiring is correct (do double check it!) it might be that you're running the chip on wrong clock speed. On the ATmega8515 datasheet it says that the L variant is capable of running at only up to 8MHz. Not 16!

It also appears to me that if you select the an external crystal as the clock from the Arduino IDE, it is impossible to communicate with the chip without a suitable crystal connected. (which I obviously didn't have)

I got around this problem by using my Arduino UNO as the ISP with a modified sketch that creates an around 1MHz clock signal on digital pin 3. I plugged this into XTAL1 pin on the 8515L and burned a bootloader successfully. I used the internal 8MHz clock from that point on since clock accuracy wasn't critical for my project.

[ArduinoISP_external1mhzclock.zip](https://github.com/MCUdude/MajorCore/files/3076714/ArduinoISP_external1mhzclock.zip)

I have included the sketch here but, I did not create it. I found it on AVRfreaks forums somewhere.

I hope this is at least somewhat helpful.

closed 2019-11-16 by MCUdude