MCUdude/MajorCore #12: atmega162 upload via bootloader problem

dsl400, 2018-06-05:
I have a few cheap VCDS clones with atmega162 mcu and ft232rl 
uploading via USBasp works, upload via serial does not work
reset via usb works

MCUdude, 2018-06-05:
Do you have the schematic for the VCDS? I guess you can't just use it without any modifications

MCUdude, 2018-06-05:
I also believe it's kind of unlikely that the VCDS have the right circuity to reset the ATmega162 properly

dsl400, 2018-06-05:
I have the schematic, I made the modifications, rx tied to tx, tx tied to rx, I made a schetch that sends data to serial and I can read serial data from the device.

void setup() {
  // put your setup code here, to run once:
  
  pinMode(33,OUTPUT);
  pinMode(32,OUTPUT);
  Serial.begin(9600);
  Serial.println("begin");
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(33,HIGH);
  digitalWrite(32,LOW);
  delay(300); 
  digitalWrite(33,LOW);
  digitalWrite(32,HIGH);
  delay(300);
  Serial.println("test");
}

I can see that the software resets whenever I open the serial monitor
however, upload via usb cable does not work
this is a schematic of the interface
https://a.d-cd.net/3c79b56s-1920.jpg

thanks for your amazingly fast reply

MCUdude, 2018-06-05:
I've had a look at the schematic. It's now very readlable, but I think I got an idea what's giong on. It seems like the circuit lacks a proper auto reset circuity. you'll need a 10k pullup resistor from RST to VCC, and a 100n capacitor wbtween DTS on the FT232 chip and RST on the ATmega162.

Can you please also post two high.res photos of you board; one from each side?
I've had a seach for it on google, but I'm having a hard time fining one that matches that exact schematic.

dsl400, 2018-06-05:
I have the resistor and capacitor in place, as I said the mcu resets every time I  open serial monitor

dsl400, 2018-06-05:
[image]
[image]

dsl400, 2018-06-05:
ATF16V8 is removed and I used wires to connect the uart

MCUdude, 2018-06-05:
> I tested clock 8Mhz internal / external and I made a 7.37MHz config
with 8MHz external serial data is broke
with 7.37MHz external serial data is ok

Can you also try to burn bootloader using the **1 MHz internal** option and see if that one works?
Did you acually compiled a 7.37MHz bootloader, or did you just edited the boards.txt file?

MCUdude, 2018-06-05:
There may infact be an issue with MajorCore/the bootloader itself, but I'm not able to test it because I don't own an ATmega162 or a device that has one it it..

I created MajorCore as a side project because I got an ATmega8515 in a development kit, and decided to add Arduino support for it. I found out that the ATmega162 was almost identical, so I added this one as well. I've done alot of testing on the ATmega8515, so I've always assumed that everything worked fine on ATmega162 as well.

BTW can you try to do a manual install of MajorCore? I've done some bootloader changes, but it's not released as a boards manager install yet.

dsl400, 2018-06-05:
burning 1MHz internal actually kills the mcu

MCUdude, 2018-06-05:
> burning 1MHz internal actually kills the mcu

Really?? That is really weird.. You can't even re-burn the bootloader? Are you using boards manager v1.0.0?

dsl400, 2018-06-05:
avrdude.exe: set SCK frequency to 1500000 Hz
avrdude.exe: error: programm enable: target doesn't answer. 1 
avrdude.exe: initialization failed, rc=-1
avrdude.exe: AVR device initialized and ready to accept instructions
avrdude.exe: Device signature = 0x000000 (retrying)
avrdude.exe: Device signature = 0x000000 (retrying)
avrdude.exe: Device signature = 0x000000
avrdude.exe: Yikes!  Invalid device signature.
avrdude.exe: Expected signature for ATmega162 is 1E 94 04

MCUdude, 2018-06-05:
Wowow!! avrdude.exe: set SCK frequency to 1500000 Hz

That's waaaay too high. You really need to slow down that programmer.

track dowm the programmers.txt file and add this:

```
usbasp_slow.name=USBasp (MajorCore)
usbasp_slow.communication=usb
usbasp_slow.protocol=usbasp
usbasp_slow.program.protocol=usbasp
usbasp_slow.program.tool=avrdude
usbasp_slow.program.extra_params=-Pusb -B32
```

Then restart Arduino IDE and choose **USBasp (MajorCore)**

dsl400, 2018-06-06:
I managed to write new fuses using avrdudess, 
and I realised that 1MHz internal will not work for me.
The board is meat to be a CAN sniffer and 9600 baud is way to slow.

should I try anything else for being able to upload firmware via USB ?

MCUdude, 2018-06-06:
> The board is meat to be a CAN sniffer and 9600 baud is way to slow.

This was just a step to make sure you could actually upload to the microcontroller, not to use for the actual application. the 1 MHz internal oscillator allways works no matter how bad the board layout are. 9600 baud also (almost) always works no matter how bad the wiring are. If you're able to program the microcontroller at 9600 baud, we might be onto something.

> PS: I can send you some boards to play with if you want

That would be great! *hansibull* is the name, and I'm using google mail.

MCUdude, 2018-07-06:
Hello again!

I've done some further investigation, and it's true that the bootloader for ATmega162 didn't even work. I'm pretty sure it's fixed now, so please do a manual install and try with your hardware.

BTW I never actually got any of the ODB2 adapters to work on my mac. Seems like there's some issues with the FTDI drivers. Anyways I was able to purchase a DIP-40 ATmega162 instead, and used my trusty old STK500 dev board instead

dsl400, 2018-07-06:
Hello Hans,

Tank you very much for your help. I will try to flash the bootlader tomorow
Tanks  Tanks  Tanks  Tanks  Tanks  Tanks

The FTDI chips are fake and maybe that is the issue.
see https://www.youtube.com/watch?v=eU66as4Bbds
I am not shure if that's the issue but it mite be.

thank you very much once again :)

dsl400, 2018-07-08:
Before I do any mistake, I wanted to ask you if I can have a go at making
the boards.txt for this new version

thanks

MCUdude, 2018-07-08:
Yes, the boards.txt file is updated 👍

dsl400, 2018-07-23:
Bootloader working. Thanks Hans!

closed 2018-07-23 by dsl400

anistor, 2019-07-24:
Hi @dsl400 ! Did you have any luck with creating a custom firmware for these VCDS clone cables? I 'm also thinking of attempting this. I'm mainly interested in creating a simple CAN monitor /logger/sniffer compatible with Lawicel protocol that seems to be used by a lot of other tools. The goal is to just create a firmware that works on these vag-com cables without any hardware mods, except maybe for the ones necessary to burn the bootloader.

EliasKotlyar, 2022-08-21:
@dsl400 @anistor Have you had any progress of repurposing the cable? I recently found out that the clones can be updated via USB alone. Would love to see arduino running on this cables, as its perfect for connecting (old) cars with unsupported odb2 protocols to torque. It could also be used as a translator to translate k-line to can or vice versa.

anistor, 2022-08-21:
Nope.

dsl400, 2022-08-21:
Hi, @EliasKotlyar @anistor, I managed to convert the device into an arduino and use it with arduino ide with the boards provided by @MajorCore. 
You will have to burn the arduino bootloader and create the reset circuit used in arduino boards ( resistor + capacitor ).
[image]

If you manage to program the atmega162 and intend to use K-line/L-line you will also have to do something about the gal16v8. Replace it with jumpers or try to undrestand the following equation

```
!o18 = !i5 AND !i9
    + !i8 AND i11
!o16 = !i4 AND !i9
    + !i7 AND i11
!o15 = !i9
!o13 = !f14
    + !i2 AND !f17 AND !i6
    + !i3 AND !f19 AND !i6
!o12 = !i2 AND !f17
    + !i3 AND !f19
```

I guess it can programmed with vcds firmware upload tool, but you loose the upload button from the arduino ide
If this subject gets more attention one could write a tool that uploads from arduino ide using the firmware update tool

anistor, 2022-08-21:
This is absolutely awesome! Thanks for sharing this @dsl400 ! I will try it out after the vacation and report how it worked.

EliasKotlyar, 2022-08-25:
hello @anistor , @dsl400 

I stopped the idea using the VCDS cable, as i dont wanted to figure out the equasion mentioned by @dsl400. However it turned out that i had a old K/D-Can Cable which was lying around and was able to make some progress with it. K-Line is working fine till now, however i am struggling with Torque support till now. Will figure it out soon i suppose.

Here is my project
https://github.com/EliasKotlyar/Subaru-SSM2-Torque-Adapter

defimad, 2024-09-19:
@dsl400 hi ive ordered some of these clones to re-purpose them as a can sniffer did you get any further with them?

dsl400, 2024-09-21:
Hi @EliasKotlyar,  I still have it around here somewhere. I used this program https://www.rubinolab.com/isolated-arduino-canbus-monitor-with-canhacker
I remember it had a quirk when selecting the port or when inserting the obd. Do not give up on the first try.

defimad, 2024-09-29:
hi @MCUdude and @dsl400  so ive been doing some experimenting i have this cable https://www.aliexpress.com/item/1005005407314750.html after many attempts at other methods including reset circuitry and software reset jumping to bootloader etc i cannot get it to upload over serial, so by randomness i decided to revert to v2.2.2 or majorcore to use the optiboot bootloader, so i burn the bootloader, and i can upload a blink sketch over serial, with no circuit modifications at all, but heres the thing it only works once, maybe it has something to do with the 16v8b, as the dtr pin from the ft232 goes to it, or maybe its something with the bootloader? any ideas? @MCUdude

dsl400, 2024-09-29:
@defimad a few years have passed, my memory about this project is now used by other projects. However I managed to find the project files and create a [repo](https://github.com/dsl400/vcds-arduino-canbus-monitor).  Hope this solves your problem

defimad, 2024-09-29:
how did you read the 16v8b because the pinout is different on this board?

dsl400, 2024-09-29:
I found the file online. The gal can not be read from what I know