MCUdude/MajorCore #20: Please help me - how to load bootloader into chip ATmega8515L

zdendazubr, 2020-02-25:
Hello,
   i would like to ask you for help, because nowhere on the web i found the answer.
I already wrote asking on:
https://forum.arduino.cc/index.php?topic=561320.0
but nobody helped me.
Problem is not solved and now I would like to use ATmega8515L to my project.

I can describe here all:

1)
In menu „File\Preferences\Additional boards manager URLs" i put new link:
       https://mcudude.github.io/MajorCore/package_MCUdude_MajorCore_index.json
In menu „Tools\Board\Boards manager" i scrolled to „MajorCore by MCUdude" and choosed „install"
In menu „Tools\Board" then i see: „MajorCore/ATmega8515". Good.
 
2)
Into Arduino UNO i loaded „File\Examples\ArduinoISP\ArduinoISP", and then i unplugged PC.
 
3)
I connected ATmega8515L on breadboard:
 
UNO (5V)    ---> ATMega8515L (pin 40 - 5V)
UNO (GND)   ---> ATMega8515L (pin 20 - GND)
UNO (pin10) ---> ATMega8515L (pin 9 - reset)
UNO (pin11) ---> ATMega8515L (pin 6 - MOSI)
UNO (pin12) ---> ATMega8515L (pin 7 - MISO)
UNO (pin13) ---> ATMega8515L (pin 8 - SCK)
 
On ATMega8515L pins 18 (XTAL2) and 19 (XTAL1) i connected crystal 16 Mhz and capacitors 22 pF to GND.
And  on Arduino UNO i used electrolyt. capacitor 10 mikrofarads between RESET-GND (minus to GND).
 
then i connect PC.
 
4)
I choose in menu:
„Tools\Board\ATmega8515"
„Tools\BOD\2.7v"
„Tools\Clock\16MHz external"
„Tools\Compiler LTO: Disabled (default)"
„Tools\Programator\Arduino as ISP"
and i tryied: „Tools\Burn bootloader"
 
But i get:
================================================================
Arduino: 1.8.3 (Windows 10), Vývojová deska: "ATmega8515, 2.7v, Disabled (default), 16 MHz external"
 
C:\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\1.0.1/avrdude.conf -v -patmega8515 -cstk500v1 -PCOM8 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0xd4:m -Ulfuse:w:0b10111111:m 
 
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch
 
         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\1.0.1/avrdude.conf"
 
         Using Port                    : COM8
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega8515
         Chip Erase delay              : 9000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :
 
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20   128    0 no        512    0      0  9000  9000 0xff 0xff
           flash         33     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
 
         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us
 
avrdude: AVR device initialized and ready to accept instructions
 
Reading | ################################################## | 100% 0.02s
 
avrdude: Device signature = 0x000000 (retrying)
 
Reading | ################################################## | 100% 0.02s
 
avrdude: Device signature = 0x000000 (retrying)
 
Chyba při vypalování zavaděče.
Reading | ################################################## | 100% 0.02s
 
avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.
 
 
avrdude done.  Thank you.
 
================================================================

I tryied use 5V / 3.3V from Arduino - unfortunatelly - same result.
I tryied use variations of BOD, Compiler LTO, 8 MHz internal crystal - unfortunatelly - same result.
I tryied put between RESET of chip and 5V rezistor - unfortunatelly - same result.
 
I normally programm ATTiny85 and ATmega328p on breadboard via Arduino ISP (only with different core) - all is ok, all chips are normally programmed (including burning bootloaders).

THANK YOU VERY MUCH FOR ANY IDEA.
Zdenda from Czech Republic

MCUdude, 2020-02-25:
Please post a photo of your current setup. That would make it much easier for me to troubleshoot. The error you're getting indicates you can't communicate with the ATmega8515. Is it brand new, or is it from some old hardware? The fuse settings might be causing this.

zdendazubr, 2020-02-25:
Hi - i am so glad for your answehr !!!
I have bought 2 pcs of ATmega8515L - both new, from electronic-schop in my country.
Here are photos:

zdendazubr, 2020-02-25:
There is not any button for attaching file - so i put photos to my web - please look here:
http://www.zzubr.cz/__ZARIDIT/atmega8515L/1.jpg
http://www.zzubr.cz/__ZARIDIT/atmega8515L/2jpg
http://www.zzubr.cz/__ZARIDIT/atmega8515L/3jpg
http://www.zzubr.cz/__ZARIDIT/atmega8515L/4jpg
http://www.zzubr.cz/__ZARIDIT/atmega8515L/5jpg
Now i can stay at my computer for about 4 hours, than i must go - so i will wait here for your reply.
Thank you for your time !

MCUdude, 2020-02-25:
There is a chance that the fuse settings have disabled the crystal driver, and thus the chip expects a driven clock signal applied to the XTAL1 pin.

Try to use [Adafruit's Arduino as ISP sketch](https://github.com/adafruit/ArduinoISP). This sketch is different from the original and adds an 8 MHz clock output on digital pin 9. Connect digital pin 9 on your UNO to the XTAL1 pin on the ATmega8515 and see if it makes a difference.

zdendazubr, 2020-02-25:
I am going to try it now.

zdendazubr, 2020-02-25:
Good news and bad news :)
Good news - it looks that bootloader is loaded.
Here is output:

C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.1/avrdude.conf -v -patmega8515 -cstk500v1 -PCOM3 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0b11010100:m -Ulfuse:w:0b10100100:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.1/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega8515
         Chip Erase delay              : 9000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20   128    0 no        512    0      0  9000  9000 0xff 0xff
           flash         33     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9306 (probably m8515)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:
[... 142 lines cut ...]
avrdude done.  Thank you.

================================================================
================================================================
Bad news is, that i next tried to load scatch - but with this output:

Projekt zabírá 924 bytů (12%)  úložného místa pro program. Maximum je 7680 bytů.
Globální proměnné zabírají 9 bytů (1%)  dynamické paměti, 503 bytů zůstává pro lokální proměnné. Maximum je 512 bytů.
C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.1/avrdude.conf -v -patmega8515 -carduino -PCOM3 -b38400 -D -Uflash:w:C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_844082/Test_ATMega8515L.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.1/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 38400
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x3b
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x3b

(last 2 lines are still repeating...)

zdendazubr, 2020-02-25:
Aha - here is COMPLETE listing:

Projekt zabírá 924 bytů (12%)  úložného místa pro program. Maximum je 7680 bytů.
Globální proměnné zabírají 9 bytů (1%)  dynamické paměti, 503 bytů zůstává pro lokální proměnné. Maximum je 512 bytů.
C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.1/avrdude.conf -v -patmega8515 -carduino -PCOM3 -b38400 -D -Uflash:w:C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_844082/Test_ATMega8515L.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.1/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 38400
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x3b
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x3b
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x3b
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x3b
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3b
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x3b
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x3b
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x3b
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x3b
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x3b

avrdude done.  Thank you.

Problém s přenosem dat na vývojovou desku (board). Na http://www.arduino.cc/en/Guide/Troubleshooting#upload naleznete další doporučení.

MCUdude, 2020-02-25:
Try to run **Burn bootloader**

zdendazubr, 2020-02-25:
I tried - output is in my comment before last comment...

zdendazubr, 2020-02-25:
no - sorry - i am stupid :) moment :))))

MCUdude, 2020-02-25:
Well, adding the clock did the trick!!

```
avrdude: Device signature = 0x1e9306 (probably m8515)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.1/bootloaders/optiboot_flash/bootloaders/atmega8515/8000000L/optiboot_flash_atmega8515_UART0_38400_8000000L.hex"
avrdude: writing flash (8192 bytes):
```

This means that you have successfully burned the bootloader to the ATmega8515. You can now disconnect the wire from pin 9 and connect the crystal and the capacitors.

zdendazubr, 2020-02-25:
When i try burn bootloader:

C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.1/avrdude.conf -v -patmega8515 -cstk500v1 -PCOM3 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0b11000100:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.1/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
avrdude: ser_open(): can't open device "\\.\COM3": Syst�m nem��e nal�zt uveden� soubor.

avrdude done.  Thank you.

Chyba při vypalování zavaděče.

MCUdude, 2020-02-25:
If you want to use the bootloader for uploading code, you will have to use a USB to serial adapter. If not, you can continue to use your Arduino UNO, but now with a 16 MHz crystal instead.

The ATmega8515 you posted a picture of is quire old, manufactured in week 14 in the year 2004. My guess is that this chip has been used by someone before you, and have sat the fuses so that it won't drive a crystal anymore. For reference, the factory fuses are set to use the internal 1 MHz oscillator, so someone must have done something with it before you.

MCUdude, 2020-02-25:
> avrdude: ser_open(): can't open device "\.\COM3"

Disconnect your UNO, reconnect and try to select the right COM port again

zdendazubr, 2020-02-25:
yes, i will:
disconnect XTAL1 from UnoPin9,
connect 16 Mhz crystal (or may i use 8Mhz internal?)
and i will try load scatch

MCUdude, 2020-02-25:
You can use the internal 8 MHz oscillator if you like. Just remember to burn bootloader again to set these settings. You can remove the crystal after you have done this.

But are you going to continue using the UNO as your programmer, or will you use a dedicated USB to serial adapter (FTDI adapter) instead)? If you're using the UNO, you will have to hold down the shift key when clicking the upload button. This will also remove the bootloader from the ATmega8515 (but that's fine, you don't need it when using a programmer).

zdendazubr, 2020-02-25:
I made all again.
1) I disconnected crystal, connected XTAL1 to UnoPin9, loaded Adafruit ArduinoISP --> output looks OK, super :)
2) next burned bootloader --> output look OK too :)
3) but than I disconnected XTAL1 from UnoPin9, connected crystal to XTAL1/XTAL2 (with 22pF capacitors) and tried to load scatch - but output is:

Projekt zabírá 924 bytů (12%)  úložného místa pro program. Maximum je 7680 bytů.
Globální proměnné zabírají 9 bytů (1%)  dynamické paměti, 503 bytů zůstává pro lokální proměnné. Maximum je 512 bytů.
C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.1/avrdude.conf -v -patmega8515 -carduino -PCOM3 -b115200 -D -Uflash:w:C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_611447/Test_ATMega8515L.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.1/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xb3
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xb3
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xb3
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xb3
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xb3
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xb3
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xb3
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xb3
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xb3
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xb3

avrdude done.  Thank you.

Problém s přenosem dat na vývojovou desku (board). Na http://www.arduino.cc/en/Guide/Troubleshooting#upload naleznete další doporučení.

zdendazubr, 2020-02-25:
now I will try all these steps but only with 8MHz internal crystal...moment please...

zdendazubr, 2020-02-25:
Unfortunatelly - same result - 10x "programmer is not responding"

MCUdude, 2020-02-25:
> but than I disconnected XTAL1 from UnoPin9, connected crystal to XTAL1/XTAL2 (with 22pF capacitors) and tried to load scatch - but output is:

Try to connect and LED to digital pin 0 (PB0). After you have burned the bootloader, the LED will blink twice every second. THis is a good indicator to check if the ATMega8515 is actually running or not.

But you haven't answered this question:
> But are you going to continue using the UNO as your programmer, or will you use a dedicated USB to serial adapter (FTDI adapter) instead)? If you're using the UNO, you will have to hold down the [...]

From the error message above it seems like you're trying to upload code using the UNO without holding down shift

zdendazubr, 2020-02-25:
Yes, I am using UNO as programator.
By the way - is correct, that I am using in menu IDE: Programmer/Arduino as ISP?

zdendazubr, 2020-02-25:
Sorry - i am going to try it with SHIFT...

zdendazubr, 2020-02-25:
I must hold SHIFT during burning bootloader OR during loading scatch?

MCUdude, 2020-02-25:
> Yes, I am using UNO as programator.
By the way - is correct, that I am using in menu IDE: Programmer/Arduino as ISP?

Yes, this is correct.

> I must hold SHIFT during burning bootloader OR during loading scatch?

You must hold down shift when loading the sketch. You can also find this option under the Sketch menu. (Sketch > Upload using programmer Ctrl+shift+U)

zdendazubr, 2020-02-25:
NOW ALL IS WORKING - BOOTLOADER AND SCATCH !!! I ABSOLUTELY LOVE YOU  !!!!!!!
May I have one question more please?

MCUdude, 2020-02-25:
> NOW ALL IS WORKING - BOOTLOADER AND SCATCH !!! I ABSOLUTELY LOVE YOU !!!!!!!

Great to hear! AVRs aren't that complicated once you know them 🙂 

> May I have one question more please?

Sure!

zdendazubr, 2020-02-25:
I found this:
https://www.tindie.com/products/MCUdude/dip-40-arduino-compatible-development-board/
it looks that it is yours :)
My question is probably stupid:
If I buy it - what will i do for this:
1) if I will want to use it to program another ATmega chip, which I will buy - how to connect it with PC for bootloading/scatchLoading? Or via Uno?
2) will I use Arduino IDE which I know, or I will must to use some another?
Thank you for answer AND I THANK YOU AGAIN FOR TIME WITH 8515 !!! THANK YOU !

MCUdude, 2020-02-25:
> I found this:
https://www.tindie.com/products/MCUdude/dip-40-arduino-compatible-development-board/
it looks that it is yours :)

That's correct, I've made this!

> if I will want to use it to program another ATmega chip, which I will buy - how to connect it with PC for bootloading/scatchLoading? Or via Uno?

With this board you can't program ATmega8515 and ATmega162, because these have a different pinout. However, I do have an adapter I can add if for free if you like so that you can use ATmega8515 and ATmega162 with it. It looks like this:
[image]

The microcontroller that comes with the board, ATmega32 already has a bootloader installed. However, if you want to use other chips you can just remove the ATmega32 from its socket and insert another one. I will, however, recommend you to buy a dedicated programmer. Personally I use [this USBtinyISP programmer](https://www.ebay.com/itm/5V-Micro-USB-Tiny-AVR-ISP-ATtiny44-USBTinyISP-Programmer/202858811063?hash=item2f3b53ceb7:g:VqIAAOSwoQdd~B0K) where I've soldered a female 2x3 header so that I can use it with all boards that have a standard, male AVRISP connector. Another very good programmer is USBasp. You can also find this on ebay (just make sure to get the 10 to 6 pin adapter with it). I recommend you to get both programmers!

> will I use Arduino IDE which I know, or I will must to use some another?

You can still use Arduino IDE with this board. All you need to do is to install the [MightyCore package](https://github.com/MCUdude/MightyCore). Note that this board has a USB to serial adapter, so you will only need a programmer if you want to change some of the settings such as internal/external clock. After you have burned the bootloader you can use it with only a USB cable, just like the Arduino UNO.

> AND I THANK YOU AGAIN FOR TIME WITH 8515 !!! THANK YOU !

You're welcome! It's always nice when we figure out the issue in the end.

zdendazubr, 2020-02-25:
Thank you, super.
By and again - thank you - it works perfect !
Zdenda

MCUdude, 2020-02-25:
Just as a note, I will be gone on vacation on Friday, so if you want to order a board, you should do it before I leave 😉

closed 2020-02-25 by MCUdude

zdendazubr, 2020-02-25:
We are now only thinking about it - in my country it is lot of money :) but if we will need this, we will buy it - it looks perfect made !

MCUdude, 2020-02-25:
That's very understandable. $40 can buy you a lot more in your country than in mine. But if you'll order it some time, just leave a note, and I'll add the DIP40 adapter (for free) so you can use the ATmega8515 with it!

zdendazubr, 2020-02-25:
Sure :) thank you and see you

zdendazubr, 2020-02-25:
OH NO - I am at home (when I wrote you before - I was in my office) and I have problem :(((
I connected:
//      ATmega8515L  -    UNO
//    ------------------------
//    pin 20         -    GND
//    pin 40         -     5V
//    pin  6 (MOSI)  - pin 11
//    pin  7 (MISO)  - pin 12
//    pin  8 (SCK)   - pin 13
//    pin  9 (RESET) - pin 10
//    pin 19 (XTAL1) - pin  9
//    rezistor 10kOhm "ATmega8515L pin 9 (RESET)" between GND on Uno
Than I loaded to Arduino UNO scatch "https://github.com/adafruit/ArduinoISP".
OK

But than I disconnected PC, added 10microfarad capacitor between RESET-GND (minus to GND).
In menu I choosed "ATmega8515/8MHz internal/2.7v/Compiler LTO: Disabled/Bootloader\Yes (UART0)/Port.../Arduino as ISP"
and tried Burn bootloader.
But it is impossible - I must to do something wrong but I do not see what :(
Please - help me again with same problem...
Here is output from burning bootloader:

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Zdenda\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.0/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega8515
         Chip Erase delay              : 9000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20   128    0 no        512    0      0  9000  9000 0xff 0xff
           flash         33     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_recv(): programmer is not responding
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.
Chyba při vypalování zavaděče.
avrdude: stk500_recv(): programmer is not responding
avrdude done.  Thank you.

MCUdude, 2020-02-25:
Open Arduino IDE settings, turn on verbose upload and post your error message again. I need to see the Avrdude command and the entire log.

zdendazubr, 2020-02-25:
I did it, here is output:

E:\_Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\Zdenda\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.0/avrdude.conf -v -patmega8515 -cstk500v1 -PCOM3 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0xd4:m -Ulfuse:w:0b10100100:m 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Zdenda\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.0/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x03

Chyba při vypalování zavaděče.
avrdude done.  Thank you.

MCUdude, 2020-02-25:
Seems like your computer can't communicate with the Arduino UNO. Try to re-upload the Arduino as ISP sketch and make sure the capacitor on the UNO is properly in place. This is why having a dedicated programmer often is better.

zdendazubr, 2020-02-25:
I closed IDE, again opened IDE, again loaded Adafruit's Arduino as ISP sketch - it looks OK - here is output:

E:\_Arduino\arduino-builder -dump-prefs -logger=machine -hardware E:\_Arduino\hardware -hardware C:\Users\Zdenda\AppData\Local\Arduino15\packages -tools E:\_Arduino\tools-builder -tools E:\_Arduino\hardware\tools\avr -tools C:\Users\Zdenda\AppData\Local\Arduino15\packages -built-in-libraries E:\_Arduino\libraries -libraries C:\Users\Zdenda\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2A03_0X0043 -ide-version=10803 -build-path C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201 -warnings=none -build-cache C:\Users\Zdenda\AppData\Local\Temp\arduino_cache_541831 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=E:\_Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=E:\_Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=E:\_Arduino\hardware\tools\avr -verbose E:\_Arduino-1.8.10\SKETCHES\Adafruit_ArduinoISP\Adafruit_ArduinoISP.ino
E:\_Arduino\arduino-builder -compile -logger=machine -hardware E:\_Arduino\hardware -hardware C:\Users\Zdenda\AppData\Local\Arduino15\packages -tools E:\_Arduino\tools-builder -tools E:\_Arduino\hardware\tools\avr -tools C:\Users\Zdenda\AppData\Local\Arduino15\packages -built-in-libraries E:\_Arduino\libraries -libraries C:\Users\Zdenda\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=0X2A03_0X0043 -ide-version=10803 -build-path C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201 -warnings=none -build-cache C:\Users\Zdenda\AppData\Local\Temp\arduino_cache_541831 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=E:\_Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=E:\_Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=E:\_Arduino\hardware\tools\avr -verbose E:\_Arduino-1.8.10\SKETCHES\Adafruit_ArduinoISP\Adafruit_ArduinoISP.ino
Using board 'uno' from platform in folder: E:\_Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: E:\_Arduino\hardware\arduino\avr
Detecting libraries used...
"E:\_Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\_Arduino\hardware\arduino\avr\cores\arduino" "-IE:\_Arduino\hardware\arduino\avr\variants\standard" "C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201\sketch\Adafruit_ArduinoISP.ino.cpp" -o "nul"
Generating function prototypes...
"E:\_Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\_Arduino\hardware\arduino\avr\cores\arduino" "-IE:\_Arduino\hardware\arduino\avr\variants\standard" "C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201\sketch\Adafruit_ArduinoISP.ino.cpp" -o "C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201\preproc\ctags_target_for_gcc_minus_e.cpp"
"E:\_Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201\preproc\ctags_target_for_gcc_minus_e.cpp"
Kompiluji projekt...
"E:\_Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\_Arduino\hardware\arduino\avr\cores\arduino" "-IE:\_Arduino\hardware\arduino\avr\variants\standard" "C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201\sketch\Adafruit_ArduinoISP.ino.cpp" -o "C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201\sketch\Adafruit_ArduinoISP.ino.cpp.o"
Compiling libraries...
Compiling core...
"E:\_Arduino\hardware\tools\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\_Arduino\hardware\arduino\avr\cores\arduino" "-IE:\_Arduino\hardware\arduino\avr\variants\standard" "E:\_Arduino\hardware\arduino\avr\cores\arduino\wiring_pulse.S" -o "C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201\core\wiring_pulse.S.o"
"E:\_Arduino\hardware\tools\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\_Arduino\hardware\arduino\avr\cores\arduino" "-IE:\_Arduino\hardware\arduino\avr\variants\standard" "E:\_Arduino\hardware\arduino\avr\cores\arduino\WInterrupts.c" -o "C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201\core\WInterrupts.c.o"
"E:\_Arduino\hardware\tools\avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IE:\_Arduino\hardware\arduino\avr\cores\arduino" "-IE:\_Arduino\hardware\arduino\avr\variants\standard" "E:\_Arduino\hardware\arduino\avr\cores\arduino\hooks.c" -o "C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201\core\hooks.c.o"
[... cut ...]
avrdude: verifying flash memory against C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201/Adafruit_ArduinoISP.ino.hex:
avrdude: load data flash data from input file C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201/Adafruit_ArduinoISP.ino.hex:
avrdude: input file C:\Users\Zdenda\AppData\Local\Temp\arduino_build_347201/Adafruit_ArduinoISP.ino.hex contains 4488 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.69s

avrdude: verifying ...
avrdude: 4488 bytes of flash verified

avrdude done.  Thank you.

zdendazubr, 2020-02-25:
Oh - I found it now - when I checked it against schema on your page:
https://github.com/MCUdude/MajorCore
I found, that 10k rezistor i have between RESET-GND.
I changed it to RESET-VCC and all is OK :)
Sorry for wasting your time !
I will not wrote you again :)))
Sorry again, thank you very much, bye !