MCUdude/MajorCore #21: Programming with your board "dip-40-arduino-compatible-development-board"

zdendazubr, 2020-06-14:
Hi again :)
I finally decided, that I will buy this very nice board:
"https://www.tindie.com/products/MCUdude/dip-40-arduino-compatible-development-board/"

But as I am total-amateur, I need some answers again.
May I please You?

1) Is that possible to buy from you beside this board also microcontrollers: 1x ATmega1284, 1x ATmega644, 1x ATmega324, 1x ATmega164, 1x ATmega32, 1x ATmega16, 1x ATmega8535 ?

2) If I will want to program these microcontrollers, I suppose to do:
      a) I will install on PC into Arduino IDE package "https://github.com/MCUdude/MightyCore"
      b) I will insert new microcontroller into this your nice board
      c) instead of Arduino Uno board I will connect only your board to PC-USB
      d) I will choose from menu corresponding menu-items for inserted microcontroller
      e) than I Burn bootloader
      f) and Sketch
Is this correct?

3) I am asking for this, because last time You wrote me:
"I will, however, recommend you to buy a dedicated programmer."
But I do not want to use another programmer - I would like to use Arduino IDE.

4) Next You wrote:
"Note that this board has a USB to serial adapter, so you will only need a programmer if you want to change some of the settings such as internal/external clock..."
So - if I will NOT want to change some of the settings, all is OK with point [2] ?

5) If I understand, there are 2 ways how to program these microcontrollers.
a) either I put them on breadboard, I use Arduino UNO board and I Program it as ISP (with your library).
b) or I put them on your board and I use direct programming (with you library) --> so not ISP.
Is this correct?

6) I would like to buy from You also 1x ATmega8515 and 1x ATmega162.
You wrote:
"With this board you can't program ATmega8515 and ATmega162, because these have a different pinout. However, I do have an adapter I can add if for free if you like so that you can use ATmega8515 and ATmega162 with it."
So I only insert one of these 2 microcontrollers into your adapter,
this adapter I will insert into your nice board,
this board I will connect to PC,
I will run Arduino IDE,
I will choose library "https://github.com/MCUdude/MajorCore" and corresponding chip,
and I will burn Bootloader and than load Sketch.
Is this process right?

7) If all is like I suppose, could You please send me total price?
I summarize all for buying:
1x https://www.tindie.com/products/MCUdude/dip-40-arduino-compatible-development-board/
1x ATmega1284
1x ATmega644
1x ATmega324
1x ATmega164
1x ATmega32
1x ATmega16
1x ATmega8535
1x ATmega8515
1x ATmega162
+ postage money

8) Do you have some board for programming this (100 pins) microcontroller?
https://www.microchip.com/wwwproducts/en/ATmega2560
I suppose that I will use library: "https://github.com/MCUdude/MegaCore".
But the board should be able to replace chips.

Thank you very much for answers
Zdenda

MCUdude, 2020-06-14:
Hi! I'll try to answer as good as I can.

> 1. Is that possible to buy from you beside this board also microcontrollers: 1x ATmega1284, 1x ATmega644, 1x ATmega324, 1x ATmega164, 1x ATmega32, 1x ATmega16, 1x ATmega8535 ?

Sorry, but I only have one of each, and I need them for testing. The only chip I have may of is the ATmega32, which comes with the board I sell. If you need these chips, I suggest you buy them from a distributor, for instance, Farnell, Mouser, Elfa or Digikey. Or Ebay if you can find them for a reasonable price.

> 2. If I will want to program these microcontrollers, I suppose to do:
a) I will install on PC into Arduino IDE package "https://github.com/MCUdude/MightyCore"
b) I will insert new microcontroller into this your nice board
c) instead of Arduino Uno board I will connect only your board to PC-USB
d) I will choose from menu corresponding menu-items for inserted microcontroller
e) than I Burn bootloader
f) and Sketch
Is this correct?

a) That's right!
b) The microcontroller is already in its socket when the board arrives. This is to prevent the pins from bending in shipping.
c) Correct
d) You choose "ATmega32" and "16 MHz external oscillator"
e) No, the bootloader is already present. All you need to do is to connect the USB cable and click upload. This is why the settings in d) is important
f) Yes, just click upload, and it will work just like an Arduino UNO

> 3. I am asking for this, because last time You wrote me:
"I will, however, recommend you to buy a dedicated programmer."
But I do not want to use another programmer - I would like to use Arduino IDE.

If you want to use another external oscillator (8, 12, 20 MHz etc.) or the internal 1, 2, 4 or 8 MHz internal oscillator, you will have to load another bootloader that can work with another frequency. You will need a programmer for this. You will also need a programmer if you want to change the BOD level. You can read more about BOD in the README in the core. So for "regular" uploads with the ATmega32 running at 16 MHz (crystal is included when you buy the board) you don't need a programmer. if you want to load a bootloader to another chip, you need a programmer. The programmer can be a USBasp, USBtinyISP or Arduino as ISP.

> 4. Next You wrote:
"Note that this board has a USB to serial adapter, so you will only need a programmer if you want to change some of the settings such as internal/external clock..."
So - if I will NOT want to change some of the settings, all is OK with point [2] ?

Yes, if you don't want to change any settings like clock and BOD, it's all OK!

> 5. If I understand, there are 2 ways how to program these microcontrollers.
a) either I put them on breadboard, I use Arduino UNO board and I Program it as ISP (with your library).
b) or I put them on your board and I use direct programming (with you library) --> so not ISP.
Is this correct?

I will assume in this question you mean ATmega8515 and ATmega162, which can be used with the board I sell if you use an adapter. This is because the ATmega8535/16/32/164/324/644/1284 has a different pinout than the ATmega8515/162. 

You can of course plug the ATmega8515/162 into a breadboard and wire it up correctly, and you should be able to either program it directly or load a bootloader.
**OR**
You can get the adapter from me for free, so you can use the ATmega8515/162 with the board I sell.

No matter what you choose to do, you will still need a programmer to program it or load the bootloader. Once the bootloader is loaded, you can upload using USB.

> 6. I would like to buy from You also 1x ATmega8515 and 1x ATmega162.
You wrote:
"With this board you can't program ATmega8515 and ATmega162, because these have a different pinout. However, I do have an adapter I can add if for free if you like so that you can use ATmega8515 and ATmega162 with it."
So I only insert one of these 2 microcontrollers into your adapter,
this adapter I will insert into your nice board,
this board I will connect to PC,
I will run Arduino IDE,
I will choose library "https://github.com/MCUdude/MajorCore" and corresponding chip,
and I will burn Bootloader and than load Sketch.
Is this process right?

Sorry, but again, I only have one ATmega8515 and one ATmega162. I need these two for testing. See my reply at 1. for a better answer on this. if you have my board and the adapter, you can just insert your ATmega8515/162, use a programmer to load the bootloader (use 16 MHz external oscillator). After the bootloader is loaded, you can upload just like you would on a "regular" Arduino. You can install MajorCore the same way as you install MightyCore. The settings in Arduino IDE is also the same.

> 7. If all is like I suppose, could You please send me total price?
I summarize all for buying:
1x https://www.tindie.com/products/MCUdude/dip-40-arduino-compatible-development-board/
1x ATmega1284
1x ATmega644
1x ATmega324
1x ATmega164
1x ATmega32
1x ATmega16
1x ATmega8535
1x ATmega8515
1x ATmega162
postage money

All I can offer is:
1x DIP-40 development board (includes ATmega32 and 16 MHz crystal in a socket)
1x adapter for ATmega8515/162

The board is priced at $40, and shipping is $6. You can have the adapter for free if you like 🙂 
Just place your order at Tindie and I will recognize your name, and make sure you get the adapter along with the board.

zdendazubr, 2020-06-14:
Thank you for answers.

1) OK, I will buy the microcontrollers on "https://cz.farnell.com".

2) I have just bought the board on tindie (name Zdenda Zubr), so please, send me the reduction for ATmega8515 and ATmega162 too.

3) Sorry for my slow thinking :) but - if I will buy for example ATmega1284 (or ATmega8515) WITHOUT burned Burnloader, I need some programmer for burning.
You wrote:
"...if you want to load a bootloader to another chip, you need a programmer. The programmer can be a USBasp, USBtinyISP or Arduino as ISP."
You mean, that I must to put this chip to breadboard and to use ISP with classic Arduino UNO connected to PC?
So - It is not possible to put it to your board and to use some your programmer?

4) And You probably overlooked my question [8] - do you have some board for programming ATmega2560 with 100 pins with CHIPS CHANGING ABILITY?
If yes, I would buy it sure !

Thank you again :)
Zdenda

MCUdude, 2020-06-14:
Thank you for your order! I'll see if I can have it shipped by tomorrow.

> 2. I have just bought the board on tindie (name Zdenda Zubr), so please, send me the reduction for ATmega8515 and ATmega162 too.

 I'll make sure it's included!

> 3. Sorry for my slow thinking :) but - if I will buy for example ATmega1284 (or ATmega8515) WITHOUT burned Burnloader, I need some programmer for burning.
You wrote:
"...if you want to load a bootloader to another chip, you need a programmer. The programmer can be a USBasp, USBtinyISP or Arduino as ISP."
You mean, that I must to put this chip to breadboard and to use ISP with classic Arduino UNO connected to PC?
So - It is not possible to put it to your board and to use some your programmer?

If you buy a new chip, it doesn't matter if it is an ATmega1284 or ATmega8515, it is completely blank. You will indeed need a programmer to either load a bootloader or load the program you've written. You don't need a breadboard to program the chip, just insert it into the board you've just ordered and use the 6-pin ISP connector. The pinout is standard and will work with all AVR ISP programmers that has six pins.

> And You probably overlooked my question [8] - do you have some board for programming ATmega2560 with 100 pins with CHIPS CHANGING ABILITY?
If yes, I would buy it sure !

Sorry, totally forgot to reply to this question. I'm sorry, but I don't have a custom board for 100-pin AVRs such as the ATmega2560. I've always been using an Arduino Mega 2560 when working with MegaCore. I'm not sure if there exists an ATmega2560 board where the chip sits on a small bard that can easily be removed.

zdendazubr, 2020-06-14:
Sorry, I do not understand - ok, I put blank new microcontroller into your board. But how should I connect this board to PC via 6-pin ISP board connector? I see that connector on your board but I can only use USB for connecting to PC. No?
(if you answer me, i will read it tomorrow morning - in my country is 0:35 am, i almost sleep...)
See you morning, thank you for time !
Zdenda

MCUdude, 2020-06-15:
> Sorry, I do not understand - ok, I put blank new microcontroller into your board. But how should I connect this board to PC via 6-pin ISP board connector? I see that connector on your board but I can only use USB for connecting to PC. No?

It's not complicated. Look at [this programmer](https://www.ebay.com/itm/143051817254). It has a USB connector in one end, and a 6-pin connector in the other. If you solder a female header to the underside of the board, You can connect it to the board you bought from me directly. By the way, most AVR based Ardunios has this 6-pin connector, so you can use this programmer on many boards.

The programmer will power the board when connected via the 6-pin connector, so you don't need to connect the USB cable.

zdendazubr, 2020-06-15:
So I must to buy this programmer module (if I want to burn Bootloder) and connect it between your board and PC ?
Or do I only solder wire with 2 ends (6-pins, and USB) and this I connect between board and PC ?

MCUdude, 2020-06-15:
> So I must to buy this programmer module (if I want to burn Bootloder) and connect it between your board and PC ?

Yes, that's correct. The tiny chip on the programmer board acts as a "translator" between USB and ISP. Again, my favorite programmers are the USBtinyISP (shown in the picture below) and the USBasp.

Here's a picture where my USBtinyISP programmer is connected to my board.
[image]

Here is a picture where the adapter is inserted.
[image]

zdendazubr, 2020-06-15:
Thank's for images.

Do you have this programmer in your shop? If yes, send me please link, I will buy it too. And You will send me it together in one package.

And when I connect it via this programmer - I will choose in Arduino IDE as programmer "USBtinyISP" and this will burn Bootloader into chip? No matter which microcontroller is insertted?

MCUdude, 2020-06-15:
> Do you have this programmer in your shop? If yes, send me please link, I will buy it too. And You will send me it together in one package.

Yes, I do. I bought mine some years ago, so the PCB is blue. For some reason, the PCBs are purple now. But that doesn't matter.

USBtinyISP programmer: https://www.ebay.com/itm/143051817254
2x3 female header: https://www.ebay.com/itm/303307459491
USBasp programmer: https://www.ebay.com/itm/232327462892

> And when I connect it via this programmer - I will choose in Arduino IDE as programmer "USBtinyISP" and this will burn Bootloader into chip? No matter which microcontroller is insertted?

Yes, these programmers are compatible with almost every AVR. Just select the microcontroller you want to program in Arduino IDE and click "Upload using Programmer" or "Burn Bootloader". If you select the wrong microcontroller, Your computer will tell you that the target is different from the one you selected.

zdendazubr, 2020-06-15:
1) I thought in YOUR SHOP - not on eBay. I thought on "**tindie**". I do not want wait whole month for board from China.

2) And please - I need your help with this.
I am choosing microcontrollers to buy but thay have more different chips in same category.
Can you please write me, **which variant of every chip WILL BE PROGRAMMABLE WITH YOUR BOARD AND LIBRARY**?
(i can buy external oscillators, if it will be necessary)

ATmega1284:
https://cz.farnell.com/microchip/atmega1284-pu/mcu-8bit-atmega-20mhz-dip-40/dp/1972067?st=ATmega1284
https://cz.farnell.com/microchip/atmega1284p-pu/mcu-8bit-atmega-20mhz-dip-40/dp/1715481?st=ATmega1284

ATmega644:
https://cz.farnell.com/microchip/atmega644p-20pu/mcu-8bit-atmega-20mhz-dip-40/dp/1455122?st=ATmega644
https://cz.farnell.com/microchip/atmega644pa-pu/mcu-8bit-avr-64k-flash-40pdip/dp/1972107?st=ATmega644
https://cz.farnell.com/microchip/atmega644a-pu/mcu-8bit-atmega-20mhz-dip-40/dp/1972105?st=ATmega644
https://cz.farnell.com/microchip/atmega644-20pu/mcu-8bit-atmega-20mhz-dip-40/dp/1288340?st=ATmega644
https://cz.farnell.com/microchip/atmega644v-10pu/mcu-8bit-atmega-10mhz-dip-40/dp/1288342?st=ATmega644

ATmega324:
https://cz.farnell.com/microchip/atmega324pv-10pu/mcu-8bit-atmega-10mhz-dip-40/dp/1455112?st=ATmega324
https://cz.farnell.com/microchip/atmega324a-pu/mcu-8bit-atmega-20mhz-dip-40/dp/1841611?st=ATmega324
https://cz.farnell.com/microchip/atmega324pa-pu/mcu-8bit-atmega-20mhz-dip-40/dp/1715485?st=ATmega324
https://cz.farnell.com/microchip/atmega324p-20pu/mcu-8bit-atmega-20mhz-dip-40/dp/1455110?st=ATmega324

ATmega164:
https://cz.farnell.com/microchip/atmega164a-pu/mcu-8bit-atmega-20mhz-dip-40/dp/1841602?st=ATmega164
https://cz.farnell.com/microchip/atmega164pa-pu/mcu-8bit-atmega-20mhz-dip-40/dp/1748515?st=ATmega164
https://cz.farnell.com/microchip/atmega164p-20pu/mcu-8bit-atmega-20mhz-dip-40/dp/1455099?st=ATmega164

ATmega32:
https://cz.farnell.com/microchip/atmega32-16pu/mcu-8bit-atmega-16mhz-dip-40/dp/9171282?st=ATmega32
https://cz.farnell.com/microchip/atmega32a-pu/mcu-8bit-atmega-16mhz-dip-40/dp/1704565?st=ATmega32

ATmega16:
https://cz.farnell.com/microchip/atmega16a-pu/mcu-8bit-atmega-16mhz-dip-40/dp/1704561?st=ATmega16
https://cz.farnell.com/microchip/atmega16l-8pu/mcu-8bit-atmega-8mhz-dip-40/dp/9171266?st=ATmega16
https://cz.farnell.com/microchip/atmega16-16pu/mcu-8bit-atmega-16mhz-dip-40/dp/9171142?st=ATmega16
https://cz.farnell.com/microchip/atmega16-16pu/mcu-8bit-atmega-16mhz-dip-40/dp/1523819?st=ATmega16

ATmega8535:
https://cz.farnell.com/microchip/atmega8535-16pu/mcu-8bit-atmega-16mhz-dip-40/dp/9171444?st=ATmega8535
https://cz.farnell.com/microchip/atmega8535l-8pu/mcu-8bit-atmega-8mhz-dip-40/dp/9171479?st=ATmega8535

ATmega8515:
https://cz.farnell.com/microchip/atmega8515l-8pu/microcontroller-mcu-8-bit-atmega/dp/2695375?st=ATmega8515
https://cz.farnell.com/microchip/atmega8515l-8pu/mcu-8bit-atmega-8mhz-dip-40/dp/9171428?st=ATmega8515
https://cz.farnell.com/microchip/atmega8515-16pu/mcu-8bit-atmega-16mhz-dip-40/dp/9171401?st=ATmega8515

ATmega162:
https://cz.farnell.com/microchip/atmega162v-8pu/mcu-8bit-atmega-8mhz-dip-40/dp/9171185?st=ATmega162
https://cz.farnell.com/microchip/atmega162-16pu/mcu-8bit-atmega-16mhz-dip-40/dp/9171169?st=ATmega162

I absoluttely do not know, which differences are among variants "P", "A", "PA", "PV", "V", and without any letter.

MCUdude, 2020-06-15:
Sorry, I don't sell any programmers. I'll recommend you to get them from China, and turn an Arduino Uno into an ISP programmer while you wait for it to arrive. There's plenty of guides on how to do this, so I won't go into more detail in this.

But why would you need all these microcontrollers? For experimenting? They are very similar, where flash and RAM sizes vary.

the "V" versions are considered low voltage versions, and if you look in the datasheet you can see that these can operate at a lower voltage than the others. However, The cannot run at 16 MHz. The "P" version is an improved version, but so is "A". It doesn't really matter too much what you get, but I'll recommend you to get the following ones if you're having all of them:

* ATmega8515-16PU
* ATmega8535-16PU
* ATmega16A-PU
* ATmega32A-PU (included with the board you bought from me)
* ATmega162-16PU
* ATmega164P or PA
* ATmega324P or PA
* ATmega644P or PA
* ATmega1284P-PU

All these can be run from 2.7 to 5.0V, and from ~0 to 20 MHz.

zdendazubr, 2020-06-15:
OK, while I will wait for programmer from China, I will put microcontrollers on breadboard and via Arduino UNO I will burn Bootloaders (as ISP). Then I will put "bootloaded" microcontrollers to your board and load sketch.

Thank You for explanation about chip-versions. I will buy those chips, which You recommended me.

Now I will wait for package. Please - do not close this issue until I will get the board - maybe I will need some help for start of using it.

Thank You very much again for now !
Zdenda

zdendazubr, 2020-07-09:
Hi - I wrote to "orders@tindie.com" but I received, that message is undeliverable with text:
"The specified email address could not be found. Please check the recipient's email address and try sending the message again."

Can you please send me some post tracking number to see where is my package now?
Thank you
Zdenda

MCUdude, 2020-07-11:
Sorry, but it has been shipped with regular airmail, so no tracking number is available. If I were to ship with a tracking number, shipping would be a little over $30. That's just too much.

I don't have any exact numbers on when it will arrive, but according to the Norwegian shipping service, it usually takes between two and three weeks.

zdendazubr, 2020-07-21:
Hi – still nothing ☹

I bought the board 14.6., so it is 5.5 weeks – may I please you – you surely have some paper from the post – could you call to the post, where is the problem with this package?
Thank you wery much.

Zdenda

Btw: I already received the microcontrollers, which you recomended me, all I bootloaded and loaded scatches via your „MajorCore“ and „MightyCore“ libraries – all works fine, You made super libraries !!!

MCUdude, 2020-07-22:
Since the board was shipped without tracking, nobody knows exactly where it is. All I know is that it will most likely show up soon. I got a message from a buyer in Belgium that ordered a board just a day after you. He got his board yesterday, after more than a month of waiting. That's unusual slow. Shipping is heavily delayed due to the Corona situation, but I don't know by how much.

zdendazubr, 2020-07-22:
> Komu: MCUdude/MajorCore <MajorCore@noreply.github.com>

Ok, so we will wait :)
By
Zdenda

Odesláno z mého chytrého telefonu Samsung Galaxy.

-------- Původní zpráva --------

zdendazubr, 2020-07-31:
Hi.
   Today i even received „Micro USB Tiny AVR ISP 5V ATtiny44 USB TinyISP Programmer Module For Arduino" from China.
As it was free shipping postage, so thay sent me it by ship, that is why it took so long time.
But your board I did not received yet – if may i please you – check, if your employee really sent it 😊
And from which country was it sent? USA?
Thank you
Zdenda

MCUdude, 2020-08-01:
> But your board I did not received yet – if may i please you – check, if your employee really sent it 😊

There are no employees here. Tindie is just a website where people like you and I can sell stuff we make ourselves. I'm assembling and selling these boards in my spare time, and the little money I make goes into my hobby; electronics, microcontrollers, and test equipment.

The board was sent from Norway, and as you've pointed out; this is taking way too long. I can refund the money you paid, but then I won't accept orders from Czech Republic for a while, because I can't afford to loose money like this several times.

zdendazubr, 2020-08-02:
Norway is much closer to Czech Rep. than China, so what the hell the post does...
As you helped me with programming chips, i take this offer like reward :)
So i will wait more time.
By
Zdenda

-------- Původní zpráva --------

zdendazubr, 2020-08-10:
Hi - I no longer hoped but in the end, today,  I received a package with a board 😊
I'm away from home today but I'll try to program it tomorrow.
By
Zdenda

MCUdude, 2020-08-10:
Great to hear it finally arrived! This must be a record, it never takes this long. Maybe it got stuck in the customs?
Anyways, I hope you find the board and the adapter useful!

zdendazubr, 2020-08-11:
Super - now I received your VERY nice board (and adapter) and also i received small USBtinyISP programmer board from China (purple one).
Now I need your help.

1) for BOOTLOADING: I insert USBtinyISP programmer into your board (with chip upside), connect only programmer to PC, and what to do? What to select in menu...

2) for loading SKETCH: I remove USBtinyISP programmer, connect your board into PC via USB and what to do?

3) power connection on your board is only for using board your alone? For bootloading and sketchloadin i do not need power connection?

Thank you for help
Zdenda

MCUdude, 2020-08-11:
* Select the microcontroller you want program. In this case it's the ATmega32
* Then select the clock frequency you want it to run on. The board is equipped with a 16 MHz external crystal in a socket that you can replace if you like. Chose either **16 MHz external** or one of the internal oscillator options.
* Select your BOD level (you can read about this in the [MightyCore readme](https://github.com/MCUdude/MightyCore#bod-option))
* Select standard pinout
* Select *Yes* if you want a bootloader, and *No* if you don't. You need a bootloader if you want to upload new programs using the USB connector on the board.
* Select USBtinyISP as your programmer
* Click Burn Bootloader

After the bootloader is loaded, you can remove the programmer and upload using the USB to serial adapter that's soldered to the board. Just remember to select the correct serial port in the tools menu. If you want to upload using the programmer for uploading, you can keep it connected and hold down the shift key on your keyboard while clicking the upload button.

The DC input jack is only needed if you want to use an external power supply that outputs more than 5V. For bootloading and sketch loading you don't need it. Just like on an Arduino UNO.

zdendazubr, 2020-08-12:
Hi.
I am trying to burn bootloader.
I inserted USBtinyISP programmer into your board.
I connected USBtinyISP programmer To PC:
[cid:image002.png@01D67088.2E4DEB90]
I set this in menu:
[cid:image001.png@01D67087.9BB16B40]
And I selected Burn bootloader.
But I get error:

=======================================================================================================================================================
Arduino: 1.8.12 (Windows 10), Vývojová deska: "ATmega32, Yes (UART0), Standard pinout, BOD 2.7V, LTO disabled, External 16 MHz"

C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5/avrdude.conf -v -patmega32 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0b11000110:m -Ulfuse:w:0b10111111:m

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: Error: Could not find USBtiny device (0x1781/0xc9f)

avrdude done.  Thank you.

Chyba při vypalování zavaděče.
=======================================================================================================================================================

MCUdude, 2020-08-12:
Are you sure you actually bought a USBtinyISP and not a USBasp programmer? Your computer can't detect it. If you open the device manager you can see what the problem is.

zdendazubr, 2020-08-12:
I bought this:

https://www.ebay.com/itm/1PC-Micro-USB-Tiny-AVR-ISP-5V-ATtiny44-USB-TinyISP-Programmer-Module-For-Arduino/143051817254?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2648

-------- Původní zpráva --------

zdendazubr, 2020-08-12:
Please DO NOT PUT on web the link from my last email. I think that login is included and enyone could get to my account to ebay :(
Sorry, i am stupid...

To be sure, i go to change password...

Zdenda

Odesláno z mého chytrého telefonu Samsung Galaxy.

-------- Původní zpráva --------

zdendazubr, 2020-08-13:
Watch please email :)

MCUdude, 2020-08-13:
Well, this is something that a forum would be better to use. The thing is that your PC can't find your programmer. You need to make sure the right driver is installed. [Install Zadig](https://zadig.akeo.ie, open, list all devices, and make sure USBtinyISP (or whatever it's called in Zadig) uses the libusb driver.

zdendazubr, 2020-08-19:
Hi.

   For now I am trying to load scatch.

(Bootloader I will try later)

I connected (to USB in PC):

[cid:image002.png@01D67601.D7DAE4A0]

I choose:

[cid:image001.png@01D67601.65AC3DC0]

Than I choose Load scatch, but I get:

Arduino: 1.8.12 (Windows 10), Vývojová deska: "ATmega32, No bootloader, Standard pinout, BOD 2.7V, LTO disabled, External 16 MHz"

C:\Arduino\Arduino-1.8.12\arduino-builder -dump-prefs -logger=machine -hardware C:\Arduino\Arduino-1.8.12\hardware -hardware C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -tools C:\Arduino\Arduino-1.8.12\tools-builder -tools C:\Arduino\Arduino-1.8.12\hardware\tools\avr -tools C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -built-in-libraries C:\Arduino\Arduino-1.8.12\libraries -libraries C:\Users\zdenek.zubr\Documents\Arduino\libraries -fqbn=MightyCore:avr:32:bootloader=no_bootloader,pinout=standard,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_904133 -warnings=none -build-cache C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_cache_833232 -prefs=build.warn_data_percentage=75 -verbose C:\Arduino\SKETCHES\test1\test1.ino

C:\Arduino\Arduino-1.8.12\arduino-builder -compile -logger=machine -hardware C:\Arduino\Arduino-1.8.12\hardware -hardware C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -tools C:\Arduino\Arduino-1.8.12\tools-builder -tools C:\Arduino\Arduino-1.8.12\hardware\tools\avr -tools C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -built-in-libraries C:\Arduino\Arduino-1.8.12\libraries -libraries C:\Users\zdenek.zubr\Documents\Arduino\libraries -fqbn=MightyCore:avr:32:bootloader=no_bootloader,pinout=standard,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_904133 -warnings=none -build-cache C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_cache_833232 -prefs=build.warn_data_percentage=75 -verbose C:\Arduino\SKETCHES\test1\test1.ino

Using board '32' from platform in folder: C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5

Using core 'MCUdude_corefiles' from platform in folder: C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5

Detecting libraries used...

"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\sketch\\test1.ino.cpp" -o nul

Generating function prototypes...

"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\sketch\\test1.ino.cpp" -o "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\preproc\\ctags_target_for_gcc_minus_e.cpp"

"C:\\Arduino\\Arduino-1.8.12\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\preproc\\ctags_target_for_gcc_minus_e.cpp"

Kompiluji projekt...

"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\sketch\\test1.ino.cpp" -o "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\sketch\\test1.ino.cpp.o"

Compiling libraries...

Compiling core...

"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles\\wiring_pulse.S" -o "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\core\\wiring_pulse.S.o"

[... cut ...]
         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5/avrdude.conf"

         Using Port                    : usb

         Using Programmer              : usbtiny

avrdude: Error: Could not find USBtiny device (0x1781/0xc9f)

avrdude done.  Thank you.

vybraný seriový port

 neexistuje a nebo váš board není připojen

It looks, that PC does not see your board ☹

I tried 2 different wires (PC – board), I tried it on 2 computers, but with same result.

What to do please?

Zdenda

MCUdude, 2020-08-19:
You will have to upload the images on the github thread rather than through Email. I can't see any of your screenshots.

zdendazubr, 2020-08-19:
I do not know how to upload image to text on github - I do not see there any button „Insert image to text from local“ ☹

And do you see them  in email?

MCUdude, 2020-08-19:
Just drag and drop the images into the text field

zdendazubr, 2020-08-19:
**OK, I am trying to type again my message, now with images :)**

Hi.
   For now I am trying to load scatch.
(Bootloader I will try later)

**I connected (to USB in PC):**
[image]

**I choose:**
[image]

**Than I choose Load scatch, but I get:**

Arduino: 1.8.12 (Windows 10), Vývojová deska: "ATmega32, No bootloader, Standard pinout, BOD 2.7V, LTO disabled, External 16 MHz"
C:\Arduino\Arduino-1.8.12\arduino-builder -dump-prefs -logger=machine -hardware C:\Arduino\Arduino-1.8.12\hardware -hardware C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -tools C:\Arduino\Arduino-1.8.12\tools-builder -tools C:\Arduino\Arduino-1.8.12\hardware\tools\avr -tools C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -built-in-libraries C:\Arduino\Arduino-1.8.12\libraries -libraries C:\Users\zdenek.zubr\Documents\Arduino\libraries -fqbn=MightyCore:avr:32:bootloader=no_bootloader,pinout=standard,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_904133 -warnings=none -build-cache C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_cache_833232 -prefs=build.warn_data_percentage=75 -verbose C:\Arduino\SKETCHES\test1\test1.ino
C:\Arduino\Arduino-1.8.12\arduino-builder -compile -logger=machine -hardware C:\Arduino\Arduino-1.8.12\hardware -hardware C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -tools C:\Arduino\Arduino-1.8.12\tools-builder -tools C:\Arduino\Arduino-1.8.12\hardware\tools\avr -tools C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -built-in-libraries C:\Arduino\Arduino-1.8.12\libraries -libraries C:\Users\zdenek.zubr\Documents\Arduino\libraries -fqbn=MightyCore:avr:32:bootloader=no_bootloader,pinout=standard,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_904133 -warnings=none -build-cache C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_cache_833232 -prefs=build.warn_data_percentage=75 -verbose C:\Arduino\SKETCHES\test1\test1.ino
Using board '32' from platform in folder: C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5
Detecting libraries used...
"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\sketch\\test1.ino.cpp" -o nul
Generating function prototypes...
"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\sketch\\test1.ino.cpp" -o "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Arduino\\Arduino-1.8.12\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Kompiluji projekt...
"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\sketch\\test1.ino.cpp" -o "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\sketch\\test1.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles\\wiring_pulse.S" -o "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_904133\\core\\wiring_pulse.S.o"
[... cut ...]
         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: Error: Could not find USBtiny device (0x1781/0xc9f)

avrdude done.  Thank you.

vybraný seriový port 
 neexistuje a nebo váš board není připojen**

It looks, that PC does not see your board ☹
I tried 2 different wires (PC – board), I tried it on 2 computers, but with same result.
What to do please?
Zdenda

MCUdude, 2020-08-19:
If you want to upload using the on-board USB, you'll have to select `Bootloader: Yes (UART0)`

zdendazubr, 2020-08-20:
So:   "**Bootloader: Yes (UART0)**"   ...means:   "**I have microcontroller already bootloaded**"
and not "**I need burn bootloader**" :)

Now board works super !!!
Afternoon, at home, I will try adapter which you sent me to load scatch to 8515.

And now please the second problem...**how to burn bootloader.**
**I bought this one:**
[image]

**In menu I have selected:**
[image]

I unplugged PC --> your board,
now (for bootloading) I plugged PC --> programmer (on your board).

But I get:

===========================================================
C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5/avrdude.conf -v -patmega32 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0b11000111:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch
         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5/avrdude.conf"
         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: Error: Could not find USBtiny device (0x1781/0xc9f)
avrdude done.  Thank you.
Chyba při vypalování zavaděče.
===========================================================

What do I wrong please?
Is ok that "Port" (in menu) is disabled?
Zdenda

MCUdude, 2020-08-20:
I have tried to explain to you several times what the problem is. Windows has not installed the proper driver for the programmer! You need to download [Zadig](https://zadig.akeo.ie). Option > List all devices. Select USBtiny, Fabisp or whatever it is called and install the libusb driver.

It does not show up as a COM port. That's the whole point.

zdendazubr, 2020-08-20:
Sorry, that I am so stupid...I do not know any "Zadig"...
But ok - I downloaded it, ran it, selected Fabisp, and installed it.
Than I tryied burn bootloader (please, do not kill me):

Arduino: 1.8.12 (Windows 10), Vývojová deska: "ATmega32, No bootloader, Standard pinout, BOD 2.7V, LTO disabled, External 16 MHz"

C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5/avrdude.conf -v -patmega32 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0b11000111:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATmega32
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    10    64    0 no       1024    4      0  9000  9000 0xff 0xff
           flash         33     6    64    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

Chyba při vypalování zavaděče.

MCUdude, 2020-08-20:
Now Avrdude are able to communicate with the USBtinyISP programmer. The problem now is that the programmer can't communicate with the target. Is the programmer plugged into the ISP header on the board? I recommend soldering female headers on the bottom side of the USBtinyISP. This makes it very easy to connect. The USB connector on the USBtinyISP should point away from the board.

What happens if you chose **USBtinyISP (slow)** instead?

zdendazubr, 2020-08-20:
OK, I will solder female headers on the bottom side of the USBtinyISP.

**Here is photo of how it is connected:**
[image]

**And, if I use  USBtinyISP (slow), I get:**

Arduino: 1.8.12 (Windows 10), Vývojová deska: "ATmega32, No bootloader, Standard pinout, BOD 2.7V, LTO disabled, External 16 MHz"

C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5/avrdude.conf -v -patmega32 -cusbtiny -B32 -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0b11000111:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
         Setting bit clk period        : 32.0
avrdude: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATmega32
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    10    64    0 no       1024    4      0  9000  9000 0xff 0xff
           flash         33     6    64    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Setting SCK period to 32 usec
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

Chyba při vypalování zavaděče.

MCUdude, 2020-08-20:
it is probably because you haven't soldered the header on the programmer. Try that first

zdendazubr, 2020-08-20:
Ok, i will try :)

Odesláno z mého chytrého telefonu Samsung Galaxy.

-------- Původní zpráva --------

zdendazubr, 2020-08-24:
**SUPER ! I soldered programmer and now I can burn bootloader - thank you !!!

And now last problem please - I am trying to load scatch to 8515 with your adapter.
I inserted chip to adapter, adapter to your board, connected only PC to your board, in menu selected this:**

[image]

**but when i loaded scatch (microcontroller already has bootloader) I get error:**

Arduino: 1.8.12 (Windows 10), Vývojová deska: "ATmega8515, Yes (UART0), BOD 2.7V, LTO disabled, External 16 MHz"

C:\Arduino\Arduino-1.8.12\arduino-builder -dump-prefs -logger=machine -hardware C:\Arduino\Arduino-1.8.12\hardware -hardware C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -tools C:\Arduino\Arduino-1.8.12\tools-builder -tools C:\Arduino\Arduino-1.8.12\hardware\tools\avr -tools C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -built-in-libraries C:\Arduino\Arduino-1.8.12\libraries -libraries C:\Users\zdenek.zubr\Documents\Arduino\libraries -fqbn=MajorCore:avr:8515:bootloader=uart0,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_692815 -warnings=none -build-cache C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_cache_75242 -prefs=build.warn_data_percentage=75 -verbose C:\Arduino\SKETCHES\test1\test1.ino
C:\Arduino\Arduino-1.8.12\arduino-builder -compile -logger=machine -hardware C:\Arduino\Arduino-1.8.12\hardware -hardware C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -tools C:\Arduino\Arduino-1.8.12\tools-builder -tools C:\Arduino\Arduino-1.8.12\hardware\tools\avr -tools C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages -built-in-libraries C:\Arduino\Arduino-1.8.12\libraries -libraries C:\Users\zdenek.zubr\Documents\Arduino\libraries -fqbn=MajorCore:avr:8515:bootloader=uart0,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_692815 -warnings=none -build-cache C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_cache_75242 -prefs=build.warn_data_percentage=75 -verbose C:\Arduino\SKETCHES\test1\test1.ino
Using board '8515' from platform in folder: C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.2
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.2
Detecting libraries used...
"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega8515 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega8515 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MajorCore\\hardware\\avr\\2.0.2\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MajorCore\\hardware\\avr\\2.0.2\\variants\\standard" "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_692815\\sketch\\test1.ino.cpp" -o nul
Generating function prototypes...
"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega8515 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega8515 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MajorCore\\hardware\\avr\\2.0.2\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MajorCore\\hardware\\avr\\2.0.2\\variants\\standard" "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_692815\\sketch\\test1.ino.cpp" -o "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_692815\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Arduino\\Arduino-1.8.12\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_692815\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Kompiluji projekt...
"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega8515 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega8515 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MajorCore\\hardware\\avr\\2.0.2\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MajorCore\\hardware\\avr\\2.0.2\\variants\\standard" "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_692815\\sketch\\test1.ino.cpp" -o "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_692815\\sketch\\test1.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\\Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega8515 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega8515 -DARDUINO_ARCH_AVR "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MajorCore\\hardware\\avr\\2.0.2\\cores\\MCUdude_corefiles" "-IC:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MajorCore\\hardware\\avr\\2.0.2\\variants\\standard" "C:\\Users\\zdenek.zubr\\AppData\\Local\\Arduino15\\packages\\MajorCore\\hardware\\avr\\2.0.2\\cores\\MCUdude_corefiles\\wiring_pulse.S" -o "C:\\Users\\ZDENEK~1.ZUB\\AppData\\Local\\Temp\\arduino_build_692815\\core\\wiring_pulse.S.o"
[... cut ...]
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x96
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x96
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x96

avrdude done.  Thank you.

Problém s přenosem dat na vývojovou desku (board). Na http://www.arduino.cc/en/Guide/Troubleshooting#upload naleznete další doporučení.

...what to do please?

MCUdude, 2020-08-24:
The ATmega8515 also needs a bootloader. You'll have to load this first. Then you can upload using the USB port on the board.

zdendazubr, 2020-08-24:
I wrote: my 8515 is already with bootloader, and scatch too - is blinking with 2 LEDs.
But ok, i will try burn bootloader with now via your board (and programmer)...moment please...
I get this:

Arduino: 1.8.12 (Windows 10), Vývojová deska: "ATmega8515, No bootloader, BOD 2.7V, LTO disabled, External 16 MHz"

C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.2/avrdude.conf -v -patmega8515 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0b11000101:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.2/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATmega8515
         Chip Erase delay              : 9000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20   128    0 no        512    0      0  9000  9000 0xff 0xff
           flash         33     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

Chyba při vypalování zavaděče.

MCUdude, 2020-08-24:
Have you inserted the adapter and chip correctly? The programmer can't communicate with the ATmega8515 for some reason. Upload a picture if you like 🙂

zdendazubr, 2020-08-24:
[image]

MCUdude, 2020-08-24:
Look correct to me. I don't remember exactly, is the ATmega8515 brand new, or have you used this chip before? What happens if you select the programmer *USBtinyISP (slow)* instead?

zdendazubr, 2020-08-24:
I bought this chip on [https://cz.farnell.com], I used it for blinking LEDs, all worked.
If I select the programmer USBtinyISP (slow) instead, I get:

Arduino: 1.8.12 (Windows 10), Vývojová deska: "ATmega8515, No bootloader, BOD 2.7V, LTO disabled, External 16 MHz"

C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.2/avrdude.conf -v -patmega8515 -cusbtiny -B32 -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0b11000101:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.0.2/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
         Setting bit clk period        : 32.0
avrdude: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATmega8515
         Chip Erase delay              : 9000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20   128    0 no        512    0      0  9000  9000 0xff 0xff
           flash         33     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Setting SCK period to 32 usec
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

Chyba při vypalování zavaděče.

MCUdude, 2020-08-24:
> I bought this chip on [https://cz.farnell.com], I used it for blinking LEDs, all worked.

Ok. Did you configure it to use internal or external crystal?

Make sure the socket is properly inserted. I'm not sure why it doesn't work for you. I did test the adapter before I shipped it to you, and it worked just fine

What you can do is wire it up like you did before you received the board from me, and then load the bootloader.

zdendazubr, 2020-08-24:
I use this:

[image]

I again tryied press to all sides, all pins looked correctly inserted.

I can burn bootloader again via another way.

Now I have to go with son to dentist,if you will have any idea, please write me and I will try it.
Thank you for interrest.

zdendazubr, 2020-08-24:
I am here, I burned bootloader with chip 8515 on breadboard and via Arduino UNO (as ISP). Than I loaded scatch (blinking LED on pin 10), all is ok.
But how to burn bootloader to 8515 on adapter on your board?

zdendazubr, 2020-08-24:
(for bootloader-burning 8515 on breadboard I use core from your library MajorCode, all is ok)

zdendazubr, 2020-08-26:
**Hi,
   ok, i will not use adapter and programming 8515 and 162. No problem.
But now I have next problem: last time I tried your board at my work and all was fine - bootloading and scatchloading.
Now I am at home and I would like to do same - bootloading and scatchloading via your board.
But nothing works... ATmega32 is normally inserted to your board, is blinking with LED on pin 10 --> bootloader is loaded (from my work) and scatch too.
But now, at home, I can not load scatch.
I have selected:**

[image]

**but i get:**

---------------------------------------------------------------------------------------
Arduino: 1.8.12 (Windows 10), Vývojová deska: "ATmega32, Yes (UART0), Standard pinout, BOD 2.7V, LTO disabled, External 16 MHz"

E:\_Arduino\Arduino-1.8.12\arduino-builder -dump-prefs -logger=machine -hardware E:\_Arduino\Arduino-1.8.12\hardware -hardware C:\Users\Zdenda\AppData\Local\Arduino15\packages -tools E:\_Arduino\Arduino-1.8.12\tools-builder -tools E:\_Arduino\Arduino-1.8.12\hardware\tools\avr -tools C:\Users\Zdenda\AppData\Local\Arduino15\packages -built-in-libraries E:\_Arduino\Arduino-1.8.12\libraries -libraries C:\Users\Zdenda\Documents\Arduino\libraries -fqbn=MightyCore:avr:32:bootloader=uart0,pinout=standard,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\Zdenda\AppData\Local\Temp\arduino_build_489005 -warnings=none -build-cache C:\Users\Zdenda\AppData\Local\Temp\arduino_cache_413634 -prefs=build.warn_data_percentage=75 -verbose E:\_Arduino\SKETCHES\test1\test1.ino
E:\_Arduino\Arduino-1.8.12\arduino-builder -compile -logger=machine -hardware E:\_Arduino\Arduino-1.8.12\hardware -hardware C:\Users\Zdenda\AppData\Local\Arduino15\packages -tools E:\_Arduino\Arduino-1.8.12\tools-builder -tools E:\_Arduino\Arduino-1.8.12\hardware\tools\avr -tools C:\Users\Zdenda\AppData\Local\Arduino15\packages -built-in-libraries E:\_Arduino\Arduino-1.8.12\libraries -libraries C:\Users\Zdenda\Documents\Arduino\libraries -fqbn=MightyCore:avr:32:bootloader=uart0,pinout=standard,BOD=2v7,LTO=Os,clock=16MHz_external -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\Zdenda\AppData\Local\Temp\arduino_build_489005 -warnings=none -build-cache C:\Users\Zdenda\AppData\Local\Temp\arduino_cache_413634 -prefs=build.warn_data_percentage=75 -verbose E:\_Arduino\SKETCHES\test1\test1.ino
Using board '32' from platform in folder: C:\Users\Zdenda\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\Zdenda\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.5
Detecting libraries used...
"E:\\_Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\Zdenda\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\Zdenda\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\Zdenda\\AppData\\Local\\Temp\\arduino_build_489005\\sketch\\test1.ino.cpp" -o nul
Generating function prototypes...
"E:\\_Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\Zdenda\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\Zdenda\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\Zdenda\\AppData\\Local\\Temp\\arduino_build_489005\\sketch\\test1.ino.cpp" -o "C:\\Users\\Zdenda\\AppData\\Local\\Temp\\arduino_build_489005\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"E:\\_Arduino\\Arduino-1.8.12\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Zdenda\\AppData\\Local\\Temp\\arduino_build_489005\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Kompiluji projekt...
"E:\\_Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\Zdenda\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\Zdenda\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\Zdenda\\AppData\\Local\\Temp\\arduino_build_489005\\sketch\\test1.ino.cpp" -o "C:\\Users\\Zdenda\\AppData\\Local\\Temp\\arduino_build_489005\\sketch\\test1.ino.cpp.o"
Compiling libraries...
Compiling core...
"E:\\_Arduino\\Arduino-1.8.12\\hardware\\tools\\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -mmcu=atmega32 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\Zdenda\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\MCUdude_corefiles" "-IC:\\Users\\Zdenda\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\variants\\standard" "C:\\Users\\Zdenda\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.0.5\\cores\\ [... cut ...]
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x9a

avrdude done.  Thank you.

Problém s přenosem dat na vývojovou desku (board). Na http://www.arduino.cc/en/Guide/Troubleshooting#upload naleznete další doporučení.
---------------------------------------------------------------------------------------

**I downloaded Zadig but when I select Menu/Options/ListAllDevices, I do not see any USBtiny or Fabisp :(
I see only:**

[image]

**please - what to do?**

MCUdude, 2020-08-27:
> ok, I will not use adapter and programming 8515 and 162. No problem.

I don't know why the adapter don't work. You can try to use a continuity tester to see which of the pins (RESET, MISO, MOSI, SCK) is not present on the ISP connector.

So you re-inserted the ATmega32 (that has a bootloader I believe), and now you can't upload to it? Are you sure it has a bootloader? Does the LED flash twice if you press the reset button?

zdendazubr, 2020-08-30:
Hi,
   1) I was trying, trying, trying and in the end - finally I was able to load scatch at home too to atmega32.
   2) but still i do not know, how to burn bootloader - still same problem - I downloaded Zadig but when I select Menu/Options/ListAllDevices, I do not see any USBtiny or Fabisp :(
I see only:
[image]
What should I do please to burn bootloader?
And again - in my work is burning ok, I have burning problem now on my PC at home...
Thank you for any idea...

MCUdude, 2020-08-31:
I have destroyed mine before. It can be done if you accidentally short out 5V and GND if your board is powered by the programmer. There is a diode on the board that goes bad. You can just remove this and solder a wire (or a 0R resistor instead

Measure the voltage across the ATtiny microcontroller. Does it have power?

zdendazubr, 2020-08-31:
If I understand correct: You write, that USB in my PC is destroyed? But I use it for external harddisks, for Arduino Uno and other devices and all is OK - I do not thing, that USB is destroyed.

MCUdude, 2020-08-31:
No, I think a diode on your USBtinyISP programmer is defective. You can check this by measuring if the 14-pin ATtiny microcontroller (that's soldered on the USBtinyISP) has 5V power or not.

[1 more reply]

closed 2023-01-17 by MCUdude