MCUdude/MajorCore #22: flash bug?

uragan1987, 2020-06-25:
hi

maybe i do understand this bootloader logic wrong

i have flashed bootloader to my atmega162 with SPI.   
FUSES are: H: 0xD2 L: 0xFF E: 0xFF, LOCK 0xFF

connected chip over RX+TX and arduino has flashed my code blink code at PE0 PIN with success
but second time to flash it no more work

is this a bug? 
do arduino overwrite the bootloader with my blink code?
do i flashed bootloader wrong? (did try over arduino and atmel studio, had success to burn it, but can only flash chip 1 time over UART)

uragan1987, 2020-06-25:
flashed file was: optiboot_flash_atmega162_UART0_115200_7372800L_B0.hex

MCUdude, 2020-06-25:
I'm currently on vacation, so I'm not able to test it on real hardware for a while. But it sounds like you have a problem with your auto-reset circuitry. Do you have a 100nF capacitor between DTR/RTS and the RST pin on the ATmega162? and do you have a 10k pullup on the RTS line?

uragan1987, 2020-06-25:
i have connected everything on stk500
and a USB to TTL to PIN PD0+PD1
DTR isnt connected
cant understand what this should do? jump to bootloader at boot?
how do you check at source code for RESET state?

MCUdude, 2020-06-25:
If you're using an STK500 board, you can't expect auto-reset to work without modifying the board.

You can find a schematic on how to wire your auto-reset circuity here:
https://github.com/MCUdude/MajorCore#minimal-setup

The capacitor pulls the reset line for a short amount of time and causes a hardware reset, and the bootloader starts.

uragan1987, 2020-06-25:
can i use 22nF?

MCUdude, 2020-06-25:
You can always try, but I can't guarantee that the capacitance it big enough that it will pull the reset line "low enough". a 220nF would be better

uragan1987, 2020-06-25:
tested with 22nF, dont work with 22nF
will buy a 100nF tomorrow and try
10k resistor is now on reset too
thy for now :)

MCUdude, 2020-06-26:
Just bear in mind that the STK500 board already has some pull-up resistor on the RST line already. You should either measure it or find its value in the schematic.

uragan1987, 2020-06-26:
ok that works fine, with 22nF, 100nF, 220nF 
havent put a pull-up resistor on the RST at STK500
my bad was, that i havent restart the chip. 
at STK500 its just hold the RESET button, 
but how to do restart atmega if chip is on a PCB?

uragan1987, 2020-06-26:
Hans thanks a lot, 
really great work

closed 2020-06-26 by uragan1987