MCUdude/MajorCore #23: bootloader flash

uragan1987, 2020-07-06:
hi

i have flashed the bootloader from compiled file: bootloader_optiboot_flash_atmega162_UART0_115200_7372800L_B0.hex
now i did a new version of the bootloader with changed LED PIN to E0

Is it posible to overwrite the current bootloader over UART0 with the new bootloader?

MCUdude, 2020-07-06:
You will have to re-upload the new bootloader using an ISP programmer.

However, I did experience some issues with the bootloader when LED flashing was activated on the ATmega162. The current hex files actually have LED flashing disabled. I don't remember exactly what the issue was about, but I thought it was a compiler bug of some sort. Anyways, feel free to try all combinations you like. The source code is there, so do what you like 🙂

uragan1987, 2020-07-07:
well done thx :)
will try to rewrite it with bootloader updater 👍

closed 2020-07-07 by uragan1987