MCUdude/MajorCore #25: MegaCore library and ATmega2560

zdendazubr, 2021-03-23:
Hi,
   may I have some questions, please?

If I buy this:
https://www.tme.eu/cz/details/atmega2560-16au/rodina-avr-8-bitu/microchip-atmel/
and this:
https://www.waveshare.com/stm32-qfp100.htm

1) will I be able to use your MegaCore library to program (and burn bootloader) this microcontroller in this adapter ?

2) is this microcontroller compatible with this adapter? (I am amateur and I do not know, if TQFP100 and QFP100 are compatible)

3) what programmer (PCB) will I need? May I use the same, what I use for programming your PCB:
https://www.tindie.com/products/MCUdude/dip-40-arduino-compatible-development-board/
which I bought ?
Thay have on web only photo with some programmer without description.

Thank you very much
Zdenda

MCUdude, 2021-03-23:
Hi!

1. You can use MegaCore to bootload the ATmega2560, but not with this adapter
2. That STM32 adapter isn't really compatible with the ATmega2560. They have different pinouts, so I can't recommend this
3. You need a standard ISP programmer to bootload the ATmega2560, for instance, a USBasp or USBtinyISP. The DIP-40 board does not bring any help or features here.

zdendazubr, 2021-03-24:
But I do not see on that adapter 40 pins - I see 100 yellow pins - watch closely please :)
And on one picture is probably seen ISP programmer - or not?

MCUdude, 2021-03-24:
But some of the 100 yellow pins may be connected together on the board, with a different connection that the ATmega2560 requires.

The picture shows a USB to serial adapter (PL2303 based) and a Segger JLINK programmer, which is not compatible with the ATmega2560. You need an ISP programmer. But why would you want this setup anyways? It's much more convenient to program than chip when it's soldered to a board, and instead use a 6-pin programming connector.

zdendazubr, 2021-03-24:
OK, I will trust You :)
And which board would you recommend me, if I need board with socket with removeable ATmega2560 chips, and with all I/O pins ready to connect to try something?

MCUdude, 2021-03-24:
The only board I know of that is designed for the ATmega2560 and has a removable socket is the [STK500 development board](https://www.microchip.com/DevelopmentTools/ProductDetails/PartNO/ATSTK500) + the old [STK504 addon board](https://cy.rsdelivers.com/product/atmel/atstk504/stk500-module-for-100-pin-lcd-avr-mega/6962487). The STK500 can be bought second hand if you're lucky, but the STK504 is very rare I believe.

MCUdude, 2021-03-24:
Oh, it seems that there are [ATmega2560 programmers with sockets at Aliexpress](https://www.aliexpress.com/item/32817634739.html?spm=a2g0o.productlist.0.0.752c6b7dUWNvCW&algo_pvid=e691cd36-59f3-4210-a03a-ba4cbabadc41&algo_expid=e691cd36-59f3-4210-a03a-ba4cbabadc41-4&btsid=0bb0623c16165790263928316e29e7&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_). You still need a programmer with a 10-pin connector, so I'll recommend USBasp here.

zdendazubr, 2021-03-24:
STK500 - i do not see on this board socket for 100 pins ATmega2560.
STK504 - looks ok, but is rare, ok.

So i will buy that one, which you found on ALiExpres.
But i know myself - I would buy definitelly something wrong - may I very please you - could you find that **board** (which you found on AliExpres) on **eBay**,
and could you find some "**USBasp programmer**" also on **eBay**,
and both links please send me and I will buy it.
Thank you very much...

And with 2 things I will only need your MegaCore library, Arduino IDE and nothing more?

MCUdude, 2021-03-24:
I'm sorry, but I can't find the 100-pin socket on Ebay, only on Aliexpress.
* Here is the USBasp programmer on eBay: https://www.ebay.com/itm/402488734484
* These adapters are nice to have for the USBasp if you want to program other boards with a 6-pin header: https://www.ebay.com/itm/173557230943

To program these you only need MegaCore and Arduino IDE, yes.

zdendazubr, 2021-03-24:
So I need **all these 3 pieces** ?

**1)** https://www.aliexpress.com/item/32817634739.html?spm=a2g0o.productlist.0.0.752c6b7dUWNvCW&algo_pvid=e691cd36-59f3-4210-a03a-ba4cbabadc41&algo_expid=e691cd36-59f3-4210-a03a-ba4cbabadc41-4&btsid=0bb0623c16165790263928316e29e7&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

**2)** https://www.ebay.com/itm/402488734484

**3)** https://www.ebay.com/itm/173557230943

Yes?

MCUdude, 2021-03-24:
You don't _need_ (3) to program the board (1), but it is very convenient to have if you want to program other AVR board, such as the board you bought from me, or an Arduino UNO. ANy board with a 6-pin ISP connector really (which is most common).

zdendazubr, 2021-03-24:
Now I understand all, I hope :)
So I will buy all 3 pieces.
**Thank you very much** for answers !
Zdenda

zdendazubr, 2021-04-22:
Hallo master,
   now, after month, I have all these pieces:
 - https://www.aliexpress.com/item/32817634739.html?spm=a2g0o.productlist.0.0.752c6b7dUWNvCW&algo_pvid=e691cd36-59f3-4210-a03a-ba4cbabadc41&algo_expid=e691cd36-59f3-4210-a03a-ba4cbabadc41-4&btsid=0bb0623c16165790263928316e29e7&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
 - https://www.ebay.com/itm/402488734484
 - ATMega2560

So - I put chip into board, connect programmer to board and to PC - and what should I choose in all menu items for bootloading please?
Thank you
Zdenda

zdendazubr, 2021-04-22:
Please - I will need more your help with this - may I?
On board are 2 slots: ISP10 and JTAG. Into which must I put that programmer?
If I use JTAG and connect it via programmer to PC, board is dark, no LED light, noactivity.
If I use ISP and connect it via programmer to PC, LED is lighting, board looks ready.
So - I installed MegaCore library to Arduino IDE, in menu I selected ATmega2560, External 8MHz, BOD 2.7V, EEPROM retained, LTO disabled, AVR pinout (???), No bootloader, Port IS DISABLED ???, and as programmer: USBtinyISP (MegaCore).

I do not know, if AVR pinout is right, I do not know why Port is in menu disabled, I do not know which MegaCore programmer should be selected.

If I use programmer: USBtinyISP (MegaCore), I get:
======================================

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch
         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/avrdude.conf"
         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: Error: Could not find USBtiny device (0x1781/0xc9f)
avrdude done.  Thank you.
Chyba při vypalování zavaděče.

If I use programmer: USBasp (MegaCore), I get:
======================================
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch
         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/avrdude.conf"
         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/
avrdude: auto set sck period (because given equals null)
avrdude: error: program enable: target doesn't answer. 1 
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.
avrdude done.  Thank you.
Chyba při vypalování zavaděče.

I absolutelly do not know, what to do - so I please You wery much for help.
Oh - and I installed USBasp via Zadig.
Could you please help me with bootloading and than loading scatch?
Thank you
Zdenda

MCUdude, 2021-04-22:
You have to remember that this is your project, not mine. I can provide missing information and/or help out fixing issues with MegaCore.

With that being said, you'll have to use the ISP port. The programmer is an USBasp, but it is usually a bit too fast for a stock ATmega2560. What happens if you try the **USBasp SLOW** programmer option? All information about bootloader options and pinouts can be found in the MegaCore README.

zdendazubr, 2021-04-22:
OH SUPER - if I try USBasp SLOW (MegaCore), it looks, that bootloader is loaded ! Thank you very much !!! First problem is probably solved - now I go try load scatch...

For information, here is result from bootloading:
-----------------------------------------------------

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 32.0
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: set SCK frequency to 16000 Hz
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: erasing chip
avrdude: set SCK frequency to 16000 Hz
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/avrdude.conf -v -patmega2560 -cusbasp -Pusb -B32 -Uflash:w:C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/bootloaders/empty/empty.hex:i -Ulock:w:0x0f:m 
Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified
[... 106 lines cut ...]
avrdude: load data flash data from input file C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/bootloaders/empty/empty.hex:
avrdude: input file C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.1.2/bootloaders/empty/empty.hex contains 0 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% -0.00s

avrdude: verifying ...
avrdude: 0 bytes of flash verified
avrdude: reading input file "0x0f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0f:
avrdude: load data lock data from input file 0x0f:
avrdude: input file 0x0f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

MCUdude, 2021-04-22:
You only need to use **USBasp SLOW** one time. The reason for this is that the default clock speed for a blank AVR is 1 MHz, and the USBasp is a little too fast for the AVR to be able to communicate with it. After you've set the settings to 8 MHz internal, you can use *USBasp* without issues. This is much faster too.

zdendazubr, 2021-04-22:
OK.
And now please - could you help me with loading scatch?
I disconnected programmer and connect board to PC directly.
1) Menu item Port is still disabled - do not you know why?
2) What sould I select in menu item Bootloader - UART0, UART1, UART2 or UART3 ?
3) Menu item Programmer is  USBasp (MegaCore)?

MCUdude, 2021-04-22:
You don't need a bootloader to load a sketch. A bootloader is only needed if you want to program the microcontroller without using a programmer, such as the USBasp. The USB port on your board is only for power, not for programming.

The board you have purchased from AliExpress is not supposed to be programmed by using a bootloader. You should use your USBasp programmer instead. The port menu is disabled because the USBasp does not show up as a serial port.

Select Bootloader: No bootloader and click upload. It will load your sketch with the USBasp programmer

zdendazubr, 2021-04-22:
Thank you wery much - this explainig is absolutelly excellent !!!
I am going to try it.

zdendazubr, 2021-04-22:
It looks, that scatch is loaded :) you are superman !!!

And may I have one last question, but which is not about MegaCore? On that board I do not see OUTPUT VCC and OUTPUT GND for testing with breadboard - or how to do it?

MCUdude, 2021-04-22:
> you are superman !!!

Hehe, absolutely not 😉 I've just been doing this for a long time.

> On that board I do not see OUTPUT VCC and OUTPUT GND for testing with breadboard - or how to do it?

By looking at the picture on AliExpress, you can probably find 5V, 3.3V, and GND on the small, yellow headers around the USB connector. Use a multimeter to measure. I'm sure it's there

zdendazubr, 2021-04-22:
Thank you wery much for all informations :)))

MCUdude, 2021-04-22:
You're welcome!

closed 2021-05-02 by MCUdude