MCUdude/MajorCore #26: Programming ATmega328P put in Arduino Uno board - but with 8MHz internal oscillator

zdendazubr, 2021-05-10:
Hi friend again !
   May I pleas you for help?
I  am trying to load scatch to ATmega328P with 8MHz internal oscillator.
This is normally working chip witch scatch, but I used it with 16MHz external oscillator.
I put this chip into Arduino Uno board,
I  set
https://mcudude.github.io/MiniCore/package_MCUdude_MiniCore_index.json
in Arduino IDE, installed it, set in menu:
- ATmega328
- Internal 8MHz       (but I tryied 4MHz, 2Mhz and 1MHz too)
- BOD 2.7V
- EEPROM retained
- LTO disabled
- 328P / 328PA
- Yes (UART0)          (but I tryied No bootloader too)
- COM3 (Arduino Uno)
And as Programmer I tryied all "xxxxx (MiniCore)" items, one by one.
But I am not able to load scatch :(
Can you help me please?
Zdenda

zdendazubr, 2021-05-10:
Hi ! :)
   I choose as Programmer: "USBasp (MiniCore)" and tryied burn bootloader.
Here is result:

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.2/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
avrdude: error: could not find USB device with vid=0x16c0 pid=0x5dc vendor='www.fischl.de' product='USBasp'

avrdude done.  Thank you.

Chyba při vypalování zavaděče.

MCUdude, 2021-05-10:
> Is correct, that I have Arduino Uno board connected to PC via USB?
> Do I not need some hardware-programmer-board?

The USB connector on the UNO is just a USB to serial adapter. You need to use a dedicated hardware programmer, such as the USBasp to but the bootloader. Just like on your ATmega2560 board.

zdendazubr, 2021-05-10:
On Arduino Uno board - there are 2 sets of 6-pins. One set 6-pins is near pin 13 and second set is near chip - which set will I need to connect, please?

zdendazubr, 2021-05-10:
Oh - I found this:
https://robu.in/burning-arduino-bootloader-with-avr-usbasp/
and there I see, that I need use 6-pins set near chip.

But now I am at work, and that programmer I have at home (I did not know, that I will need it) - now Is here 9:35 am, I will try it at home after 6:00pm.
Thank you very much for now !

zdendazubr, 2021-05-10:
Hi. I am at home now...
If I use small USBasp programmer board, I burn bootloader - that is great !
But if I try to load scatch, I must wait very long time (minutes) and get this:

Projekt zabírá 1108 bytů (3%)  úložného místa pro program. Maximum je 32256 bytů.
Globální proměnné zabírají 11 bytů (0%)  dynamické paměti, 2037 bytů zůstává pro lokální proměnné. Maximum je 2048 bytů.
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xb7
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xb7
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xb7
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xb7
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xb7
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xb7
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xb7
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xb7
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xb7
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xb7
Problém s přenosem dat na vývojovou desku (board). Na http://www.arduino.cc/en/Guide/Troubleshooting#upload naleznete další doporučení.

I tryied Zadig and reinstal USDasp driver, but I still get the same result :(
Please - what to do for scatch loading?

zdendazubr, 2021-05-11:
Oh - if I click "Upload using Programmer", then scatch is loaded :)
And thank you for advices.
Zdenda

zdendazubr, 2021-05-11:
I am not sure, that you receive these my last comments ?
Yesterday, when I added comment here, or you added comment here, I get email with copy.
Today I did not get any email and you do not answehr - I thing, that something is wrong - may be with my account ?
May I please you - write here one easy comment for check?
Thank you
Zdenda

MCUdude, 2021-05-11:
There is nothing wrong with your account, it's just that I have limited time, and most of the things you need help with figuring out are things you can easily find answers to online.

The bootloader is used so you can upload using a USB to serial interface, which is the USB port on the Arduino UNO.
You can also upload the sketch using a programmer, and you use the programmer to set fuses and burn the bootloader.

Are you sure you have a bootloader installed? If you press the reset button, the LED on digital pin 13 should flash twice within a second.

Also, when you paste your error messages you leave out the most useful part - the Avrdude command itself. Open your IDE settings and enable verbose upload. This will reveal the Avrdude command too.

zdendazubr, 2021-05-11:
I thing, that I have now correctly burned bootloader and correctly loaded scatch.
And I hope, that now is chip working with 8MHz internal oscillator and will not need external.

Just for assurance - can you check my outputs for bootloader and for scatch?

Here is for Bootloader:
===============

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.2/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.2/avrdude.conf -v -patmega328p -cusbasp -Pusb -Uflash:w:C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.2/bootloaders/optiboot_flash/bootloaders/atmega328p/8000000L/optiboot_flash_atmega328p_UART0_38400_8000000L_B5.hex:i -Ulock:w:0x0f:m 
Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

[... 177 lines cut ...]

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_44036/test1.ino.with_bootloader.hex"
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% 0.41s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_44036/test1.ino.with_bootloader.hex:
avrdude: load data flash data from input file C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_44036/test1.ino.with_bootloader.hex:
avrdude: input file C:\Users\ZDENEK~1.ZUB\AppData\Local\Temp\arduino_build_44036/test1.ino.with_bootloader.hex contains 32768 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.27s

avrdude: verifying ...
avrdude: 32768 bytes of flash verified

avrdude done.  Thank you.

MCUdude, 2021-05-11:
You're still leaving out the important Avrdude command. It's supposed to look like this:
```
C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.2/avrdude.conf -v -patmega328p -cusbasp -Pusb -Uflash:w:C:\Users\zdenek.zubr\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.2/bootloaders/optiboot_flash/bootloaders/atmega328p/8000000L/optiboot_flash_atmega328p_UART0_38400_8000000L_B5.hex:i -Ulock:w:0x0f:m 
```

You don't have to use your USBasp programmer to upload your sketch after you've burned the bootloader. You can use the on-board USB connector on your UNO board. Does the LED flash twice when you press the reset button?

zdendazubr, 2021-05-11:
OK :) I do not want to try your patience :)
Thank you for all advices.
By friend !

closed 2023-01-17 by MCUdude