MCUdude/MajorCore #28: ATmega8515 I2C

ITstreet1, 2022-10-21:
I just walk through the datasheet. There is no I2C for this MCU, or I am wrong?

And an additional question, by using your package support, can I burn the bootloader and use it just like any other ATmega MCU, say 328p?

MCUdude, 2022-10-21:
The ATmega8515 (and ATmega162) does not have i2c support.

> And an additional question, by using your package support, can I burn the bootloader and use it just like any other ATmega MCU, say 328p?

Yes, the Optiboot bootloader works exactly as you would expect from an ATmega328P.

ITstreet1, 2022-10-21:
Then, I believe I can use some softwire lib, but which pins should be suitable?

MCUdude, 2022-10-21:
> Then, I believe I can use some softwire lib, but which pins should be suitable?

Sure, but that's not up to me to decide. The ATmega8515 is very old and does not have PCINTs on the ports. ATmega162 is newer, but both of these are not very common, so I doubt you'll find an Arduino library that will work for these.

One thing you can try is to port the TinyWire library used with ATtiny13/MicroCore:
https://github.com/MCUdude/MicroCore/tree/master/avr/libraries/TinyWire

MCUdude, 2022-10-21:
A quick Google search reveals that this library is generic and platform-independent. It may work with ATmega8515.

https://github.com/stevemarple/SoftWire

ITstreet1, 2022-10-21:
That one was in my mind.
Thank you for your time. I just want to give it a try for an MCU I have.
I will post a schematic later, just to have someone check if I made a thing or two wrong.

ITstreet1, 2022-10-21:
[Schematic_Clicker ATmega 8515_2022-10-21.pdf](https://github.com/MCUdude/MajorCore/files/9840129/Schematic_Clicker.ATmega.8515_2022-10-21.pdf)
Something like this. ;)

MCUdude, 2022-10-21:
The schematic is a bit messy and hard to read but appears to be OK. I'm sure it will work as long as the schematic symbols are correct!

ITstreet1, 2022-10-22:
It is messy. It is half an hour design. LOL.
It should use MikroE mikroBUS. As for I2C, I used two sets of pins just in case one won't work.

ITstreet1, 2022-12-21:
OK,
So, it is soldered but stuck on the bootloader. I connected Nano with this one.

Nano - 8515

5V - 5V
GND - GND
D10 - RST
D11 - MOSI/PB5
D12 - MISO/PB6
D13 - SCK/PB7

- Double-checked all connections. The wires are new and tight.
- On Nano, I uploaded ArduinoISP.
- Selected 8515 of yours, port, etc
- Programmer - Arduino as ISP
- Burn bootloader

I get:

Arduino: 1.8.19 (Windows 10), Board: "ATmega8515, Yes (UART0), EEPROM retained, BOD 2.7V, LTO disabled, External 16 MHz"

avrdude: Version 6.3-20201216

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\dekip\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : COM14

         Using Programmer              : stk500v1

         Overriding Baud Rate          : 19200

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x15

avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x15

         AVR Part                      : ATmega8515

         Chip Erase delay              : 9000 us

         PAGEL                         : P00

         BS2                           : P00

         RESET disposition             : dedicated

         RETRY pulse                   : SCK

         serial program mode           : yes

         parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         ByteDelay                     : 0

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                  Block Poll               Page                       Polled

           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           eeprom         4    20   128    0 no        512    0      0  9000  9000 0xff 0xff

           flash         33     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff

Error while burning bootloader.

           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00

           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00

           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500

         Description     : Atmel STK500 Version 1.x firmware

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x02

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10

         Hardware Version: 4745024

         Firmware Version: 0.4611299

         Topcard         : STK502

         Vtarget         : 1.8 V

         Varef           : 0.0 V

         Oscillator      : Off

         SCK period      : 0.1 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01

avrdude: initialization failed, rc=-1

         Double check connections and try again, or use -F to override

         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ITstreet1, 2022-12-21:
UPDATE!!!!

I found a solution here:
https://forum.arduino.cc/t/avrdude-stk500_disable-protocol-error-expect-0x14-resp-0x10/590824

Besides, as I used Nano SPI lines from headers, rather than from ICSP, I uncommented line 81 ->  #define USE_OLD_STYLE_WIRING

Also, I added a 10uF cap on RST and GND on Nano. The bootloader is done:

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\dekip\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : COM14
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega8515
         Chip Erase delay              : 9000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20   128    0 no        512    0      0  9000  9000 0xff 0xff
           flash         33     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9306 (probably m8515)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
[... 119 lines cut ...]
avrdude: load data flash data from input file C:\Users\dekip\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.1.3/bootloaders/optiboot_flash/bootloaders/atmega8515/16000000L/optiboot_flash_atmega8515_UART0_115200_16000000L_B0.hex:
avrdude: input file C:\Users\dekip\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.1.3/bootloaders/optiboot_flash/bootloaders/atmega8515/16000000L/optiboot_flash_atmega8515_UART0_115200_16000000L_B0.hex contains 8192 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 8192 bytes of flash verified
avrdude: reading input file "0x0f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0f:
avrdude: load data lock data from input file 0x0f:
avrdude: input file 0x0f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

ITstreet1, 2022-12-21:
UPDATE 2!!!!

I can not upload a simple Blink sketch on 8515. I get:

Arduino: 1.8.19 (Windows 10), Board: "ATmega8515, Yes (UART0), EEPROM retained, BOD 2.7V, LTO disabled, External 16 MHz"

Sketch uses 890 bytes (11%) of program storage space. Maximum is 7680 bytes.

Global variables use 9 bytes (1%) of dynamic memory, leaving 503 bytes for local variables. Maximum is 512 bytes.

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x30

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x30

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x30

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x30

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x30

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x30

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x30

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x30

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x30

avrdude: stk500_recv(): programmer is not responding

avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x30

Problem uploading to board.  See https://support.arduino.cc/hc/en-us/sections/360003198300 for suggestions.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

MCUdude, 2023-01-17:
It's most likely a hardware issue. Are you trying to upload your code using your Arduino as ISP or using the onboard USB to serial adapter?

If you're using the ISP programmer, go to Sketch >Upload Using Programmer in the Arduino IDE menu

ITstreet1, 2023-01-17:
Ok, I burned the bootloader again using Nano. I have an LED on pin D0(PB0) which started blinking after the bootloader is burned. More like two fast blinks, than a sec delay.
After restarting, it stopped. No simple blink sketch, all wires the same, nothing pulled out.

Case 1:
Nano as ISP programmer.

> 
> avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x15
> avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x15
> 
> avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14
> 
> avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x01
> avrdude: stk500_initialize(): (a) protocol error, expect=0x14, resp=0x10
> avrdude: initialization failed, rc=-1
>          Double check connections and try again, or use -F to override
>          this check.
> 
> avrdude: stk500_disable(): unknown response=0x12
> the selected serial port avrdude: stk500_disable(): unknown response=0x12
>  does not exist or your board is not connected

Case 2:
Nano as ISP programmer with a CAP on RST and GND.

> avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x15
> avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x15
> 
> avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14
> 
> avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x01
> avrdude: stk500_initialize(): (a) protocol error, expect=0x14, resp=0x10
> avrdude: initialization failed, rc=-1
>          Double check connections and try again, or use -F to override
>          this check.
> 
> avrdude: stk500_disable(): unknown response=0x12
> the selected serial port avrdude: stk500_disable(): unknown response=0x12
>  does not exist or your board is not connected

Case 3:
Using onboard CP2102.

It holds on Uploading... for a long time. Meanwhile LED on D0(PB0) is blinking the same way as I mentioned above. After a while it throws:

- avrdude: stk500_recv(): programmer is not responding
- avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x16
- avrdude: stk500_recv(): programmer is not responding
- avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x16
- avrdude: stk500_recv(): programmer is not responding
- avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x16
- avrdude: stk500_recv(): programmer is not responding
- avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x16
- avrdude: stk500_recv(): programmer is not responding
- avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x16
- avrdude: stk500_recv(): programmer is not responding
- avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x16
- avrdude: stk500_recv(): programmer is not responding
- avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x16
- avrdude: stk500_recv(): programmer is not responding
- avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x16
- avrdude: stk500_recv(): programmer is not responding
- avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x16
- avrdude: stk500_recv(): programmer is not responding
- avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x16
- Problem uploading to board.  See https://support.arduino.cc/hc/en-us/sections/360003198300 for suggestions.

Settings are external 16MHz, BOD 2.7V, EEPROM retained, LTO disabled, and Bootloader YES(UART0). The programmer is left selected Arduino as ISP if that matters.

What else should I try?

Offtopic...
I have almost the same issues on ATemaga328p, but the issue is closed. :(
https://github.com/MCUdude/MiniCore/issues/215#issuecomment-1365378976

MCUdude, 2023-01-17:
Well, I don't know what more I can do. The Optiboot bootloader is proven, and I have working boards with both the ATmega8515 and ATmega328PB and Optiboot works perfectly fine. You have custom-designed boards, and neither of yours works. I'm very sure it's hardware related, but since I don't have your hardware, there's little I can do other than look at your schematic.

But does the USB to serial adapter work? Does the echo sketch below work when you upload the program using your ISP programmer?

```c
int incomingByte = 0;    // for incoming serial data

void setup() {
    Serial.begin(9600);    // opens serial port, sets data rate to 9600 bps
}

void loop() {
  // send data only when you receive data:
  if (Serial.available() > 0) {
  
    // read the incoming byte:
    incomingByte = Serial.read();
  
    // say what you got:
    Serial.print((char)incomingByte);
  }
  
}
```

ITstreet1, 2023-01-18:
How to check the communication between CP2102 and 8515?
On 8515 I can not upload any sketches.

MCUdude, 2023-01-18:
> How to check the communication between CP2102 and 8515?

Upload the test sketch I posted using your ISP programmer.
Open the serial port at 9600 baud, and try to send something. If what you sent gets printed on the screen, it means the microcontroller receives the data correctly and sends it back correctly.

ITstreet1, 2023-01-18:
On 8515 I can not upload anything. Except for the bootloader.

MCUdude, 2023-01-18:
> On 8515 I can not upload anything. Except for the bootloader.

Yes, you can. You can upload the sketch using your programmer. 

https://docs.arduino.cc/hacking/software/Programmer

closed 2023-01-19 by ITstreet1