MCUdude/MajorCore #29: Wrong signature for 8515

KOBRA18, 2023-07-13:
I don't know what is wrong. It says that device signature is probably 8515 but then error says it should be different. I tried on multiple 8515 chips and always same error. I tried changing different crystals. It worked when I uploaded bin file from MajorCore/avr/libraries/AVR_examples/examples/Blink via AVRDUDESS. This error comes if I want to erase bootloader or upload using programmer in Arduino.
Wiring is same as minimal setup, I just added 10uF cap betwen rst and gnd.

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\Username\AppData\Local\Arduino15\packages\MajorCore\hardware\avr\2.2.1\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 32.0
         AVR Part                      : ATmega8515
         Chip Erase delay              : 9000 us
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    20   128    0 no        512    1      0  9000  9000 0xff 0xff
           flash                  33     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: set SCK frequency to 16000 Hz
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9301 (probably 8515)
avrdude main() error: expected signature for ATmega8515 is 1E 93 06
        double check chip or use -F to override this check

avrdude done.  Thank you.

Failed chip erase: uploading error: exit status 1

MCUdude, 2023-07-13:
Hi!

MajorCore only supports the *ATmega8515* and not the AT90S8515, as this is an older part that's not in production anymore.

The AT90S8515 may work with a hex file compiled for the ATmega8515, but I can't guarantee anything. You can upload any hex file you want to the AT90S8515 using Avrdude directly (what I'm usually doing when not using Arduino IDE) or by using Avrdudess.

KOBRA18, 2023-07-13:
That makes sense. I really have AT90S8515. I hoped they can be brought back to life since I only know how to program in Arduino.
Thanks for all the help.

closed 2023-07-13 by KOBRA18