MCUdude/MajorCore #30: problem with Atmega162

digi55, 2024-02-25:
Hi,
Thank you for such a library.
I am using Arduino ide v2.3.2.
I detected a problem on Atmega162 in Major core 2.1.3 and above.
And also Including 3.0.1, this problem exists.
When I use Atmega162 with Pin32 (INT2) interrupt, the program is reset and restarts from the beginning.
This problem does not exist in Major Core 2.1.2.
Best Regards.

MCUdude, 2024-02-25:
Can you provide a bare minimum sketch so I can try to reproduce the issue?

MCUdude, 2024-02-25:
Thanks! I'm not at my computer at the moment, but I'll look into it later. Thanks for reporting!

MCUdude, 2024-02-25:
BTW it's really weird that it works with 2.1.2, but not 2.1.3.

If you look at what changes were made in 2.1.3 compared to 2.1.2, none of them are related to the interrupt functionality...
https://github.com/MCUdude/MajorCore/compare/v2.1.2...v2.1.3

digi55, 2024-02-25:
I think it is very interesting, There is no problem with 2.1.2. Everything is ok.
In all higher versions, including 3.0.1, the program is reset when int2 is triggered.

digi55, 2024-03-02:
Hi, have you had a chance to look at the problem?

digi55, 2024-06-28:
Hi, @MCUdude
The problem of Atmega162 resetting when INT2 is triggered, also continues in v3.02
I submit it for your information
best regards.